meshtastic/firmware #6566: [Bug]: BST not calculated correctly for POZIX TZ "GB" on 2.6.4

aghl100, 2025-04-11:
### Category

Other

### Hardware

Heltec V3

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4

### Description

Having installed 2.6.4 on a HeltecV3 and configured it for POZIX timezone "GB", the time displayed on the OLED is GMT rather than BST i.e. one hour behind. I'm not sure if the time was then picked up from USB connection to the machine which flashed it or BT connection to my Android mobile (running the app and supplying GPS to the Heltec), but I've checked the time settings on both of these are correct. 

### Relevant log output

```Shell

```

Talie5in, 2025-04-12:
You need to use the correct input  - see https://meshtastic.org/docs/configuration/radio/device/#tzdef-timezone-definition

So for BST "GMT0BST,M3.5.0/1,M10.5.0" is what is entered into the field (eg, for London)

aghl100, 2025-04-13:
Thank you very much, and sorry for not spotting that in the docs. That's fixed the issue.

closed 2025-04-13 by aghl100