meshtastic/firmware #6571: [Bug]: Location not precise although selected

SR7673, 2025-04-12:
### Category

Other

### Hardware

T-Beam

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.5

### Description

The location shown is not precise, although i selected precise location in the channel settings. It is not a GPS reception issue. Even a node with fixed location entry does not show its precise location. 
Also observed on a RAK4631

### Relevant log output

```Shell

```

kingoffoxes, 2025-04-14:
Also getting this issue on T-beam 2.6.5, checked settings on iOS app, map shows larger precision circle around rough node location.

madeofstown, 2025-04-14:
Setting precise locations on default public channels is not allowed on iOS. If you want precise locations on the default public channel you need to change the setting from an Android phone or the python command line utility.

SR7673, 2025-04-15:
I my case, i observerd this in a private and primary channel. Android was used...

kingoffoxes, 2025-04-16:
In my case the primary channel is private and the secondary channel is public. 
I am also noticing this on other firmware versions. ie 2.5.22. connecting via the web client shows that precise location is set.

luisfromero, 2025-06-05:
In my case, GPS position reports have an error of 20 to 50 meters, even in open sky and ideal GPS conditions. This is far beyond acceptable for location tracking.

When I use a non-Meshtastic firmware I found online — t-echo-lora-aprs-with-sd-v1.1.uf2 — under exact same conditions, the GPS error improves drastically to 1–5 meters, which is expected for u-blox modules under clear skies.

This suggests the hardware is not at fault and the issue is specific to the Meshtastic firmware.

According to DeepSeek, the variable precision in allocPositionPacket() should be private, as it might be modified by another channel. I'm not sure whether this is directly related to the GPS inaccuracy...

closed 2025-07-29 by github-actions[bot]