meshtastic/firmware #6577: [Bug]: Power telemetry not reporting on linux native / meshtasticd

dswinton, 2025-04-13:
### Category

Other

### Hardware

Linux Native

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.5.14

### Description

I'm having issues getting power readings from an INA219 over I2C on Meshtasticd / Linux Native.
It seems that the sensor is recognised, but then no readings are sent out to the mesh.

Testing nodes:
* Hardware is a Femtofox (running v1.0.0 alpha of Foxbuntu)
* I2C is enabled and I'm able to get power readings from the sensor within linux (not via Meshtastic)
* During startup, Meshtasticd recognises i2c devices.. so it can definitely see it
* I2C ID is 0x40, which is supported by meshtastic
* Power telemetry is enabled and set to 30 minutes
* No readings or relevant errors appear in the logs.. after acknowledging that it sees i2c sensors, the expected log entries with power telemetry readings never appear

When I test the same sensor connected to an NRF board rather than meshtasticd I do get readings over the mesh, and serial log entries with the readings...  so I'm confident in the hardware and configuration.

I had read that not all modules are enabled on Linux Native... is it possible that the power telemetry module is not enabled / supported?

If it's not enabled then... feature request.. could it be enabled, please? 😄 

### Relevant log output

```Shell

```

KyruJames, 2025-04-14:
Same here on 2.6.4. 

Meshtastic log shows it discovered at 0x40, i can read it through linux.

Nothing on Meshtastic.

madeofstown, 2025-04-14:
Power telemetry is not enabled or I've never been able to get it to work. I believe environmental sensors don't work either.

dwhacks, 2025-04-19:
I ran into this also. Would be rad if the documentation on linux native mentioned that sensors aren't enabled. Seems like the only thing that works in the oled display.

dswinton, 2025-04-21:
Heya @jp-bennett - sorry for the ping, but hoping you can share some of your wisdom here! 🙏🏻 

What was the blocker on enabling sensors/telemetry on Linux Native?

I've started having a bit of a dig into it but i can't find any leads on what needs to be addressed to get it working.

Would appreciate your insight here!  I'm happy to have a go at it, wanted to check in with you first in case there is an underlying issue I'm not aware of!

Cheers 🍻

jp-bennett, 2025-04-21:
> Heya [@jp-bennett](https://github.com/jp-bennett) - sorry for the ping, but hoping you can share some of your wisdom here! 🙏🏻
No problem! This has been on my todo list for a long time. I have a WIP branch that I'll push up in a bit.
> 
> What was the blocker on enabling sensors/telemetry on Linux Native?
The main blocker I ran into was that many of the Arduino libs refused to compile under Portduino. And because portduino was hard-coded not to use sensors, there's a huge ball of wax to untangle, to make it work.

See https://github.com/meshtastic/firmware/pull/6646 and https://github.com/meshtastic/framework-portduino/pull/55

jp-bennett, 2025-06-14:
I think this may be working now.

closed 2025-06-14 by thebentern