meshtastic/firmware #6596: [Feature Request]: Do not provide an ack when connected to mqtt when only uplink is enabled

RCGV1, 2025-04-15:
### Platform

Cross-Platform

### Description

Often I like being connected to my node and have mqtt client proxy but it makes it impossible to send stuff out to the mesh because I do not know if my messages made it as they are all acknowledged and the 3 times retry logic does not kick in and makes my messages not make it out most of the time. If downlink is not enabled, do not ack messages to mqtt as that shows the user is not using mqtt to chat through, or make it so the retries are not sent to the mqtt server.

GUVWAF, 2025-04-20:
Are you sure this is the case? I just tested this and it doesn't happen for me. 

You only get the ACK when you receive the packet from yourself when you're subscribed to the topic:
https://github.com/meshtastic/firmware/blob/5d48d2c0a74b1a561246899e7786aa1460cf0ef1/src/mqtt/MQTT.cpp#L62-L67

When downlink is not enabled, we will not subscribe to the topic and hence you will not get the ACK:
https://github.com/meshtastic/firmware/blob/5d48d2c0a74b1a561246899e7786aa1460cf0ef1/src/mqtt/MQTT.cpp#L528-L532

It might be you have to reboot when you changed the channel settings (downlink), as subscribing (or not) is done only once after booting.

RCGV1, 2025-04-20:
Test with mqtt client proxy and uplink, you get an implicit ack

GUVWAF, 2025-04-20:
Then it sounds like the client is subscribing to the topic even if you don't have downlink enabled, which seems unnecessary.

closed 2025-06-13 by github-actions[bot]