meshtastic/firmware #6606: [Bug]: Wifi does not work on M5 Stack

Basti77, 2025-04-16:
### Category

WiFi

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4  and Greater

### Description

Unfortunately the Wifi does not work on the M5 Cube.
This is a pity because I have a working 1W 433Mhz module for this hardware. 
The Compact Cube is theoretically a perfect MQTT gateway, but without Wifi you can forget it. 

### Relevant log output

```Shell

```

fifieldt, 2025-04-17:
Hi, thanks for writing using Meshtastic :)

I'm not familiar with an M5 Cube and it was difficult to google for, can you provide a link with information about it?

Also, when you mention "Wifi does not work", what behaviour do you see exactly?

Basti77, 2025-04-17:
Thanks for your reply:

There are ready‑made builds you can flash with Meshtastic for this modular hardware.
I actually quite like the hardware, because it offers a large colour display, a buzzer, three buttons, and an SD‑card slot.
https://eckstein-shop.de/M5StackESP32BasicCoreIoTDevelopmentKitArduino2FMicroPython2FUIFlow

I originally bought it to use as an APRS tracker.
https://forum.aprs-dl.de/showthread.php?tid=31
For that you had to add a LoRa shield yourself—rewarded with a full 1 W of transmit power.

Inside my units I’ve also tucked a 1,500 mAh Li‑Po and a GPS module; I don’t think you can get much more compact than that.

[image]

If you configure Wi‑Fi in Meshtastic, the device never connects over it (unlike, say, a Heltec board).
In that situation the only way to reach the board is via USB.

Unfortunately, I don’t know how to obtain any further debug information.

fifieldt, 2025-04-17:
Thanks for the great information.

It looks like the definition for that device is here 
https://github.com/meshtastic/firmware/blob/master/variants/m5stack_core/variant.h

It might be a simple matter of adding HAS_WIFI in there somewhere, but hard to tell without having one.

Could you plug in your device with USB to a computer then go to https://flasher.meshtastic.org and click on Open Serial monitor, and send us logs from there? The first minutes after startup are particularly useful as are any after you enable or disable wifi.

Basti77, 2025-04-17:
`All1243
Info302
Debug764
Warn33
Error12
Disconnect

�L��J����JH9�ό)Ί��-��.͈T)��EHQ��n9�9)���H����J��M

>�� Ϋ!�!���9�B���)](?�9���'�0��j�N|���0�1�r

�D��mΈd9)N���H�Nsʸ��p�����

��)�$ĕ�L9)��Q�nJ=���t��p!6�9� �"���������h��hr1���DEBUG | ??:??:?? 3 [GPS] Trying $PCAS06,1*1A (ATGM33xx Family)...

DEBUG | ??:??:?? 3 [GPS] Trying $PAIR021*39 (Airoha Family)...

DEBUG | ??:??:?? 4 [GPS] Trying $PQTMVERNO*58 (LC86)...

DEBUG | ??:??:?? 5 [GPS] Trying $PCAS06,0*1B (L76K)...

DEBUG | ??:??:?? 5 [GPS] Trying $PMTK605*31 (MTK Family)...

WARN | ??:??:?? 6 [GPS] No GNSS Module (baudrate 9600)

INFO | ??:??:?? 6 [ExternalNotification] Turning off external notification:

INFO | ??:??:?? 6 [ExternalNotification] 0

INFO | ??:??:?? 6 [ExternalNotification] 1

INFO | ??:??:?? 6 [ExternalNotification] 2

INFO | ??:??:?? 6 [ExternalNotification]

INFO | ??:??:?? 6 [RangeTest] Range Test Module - Disabled

INFO | ??:??:?? 6 [PowerFSM] Init the NimBLE bluetooth module

INFO | ??:??:?? 7 [Screen] Done with boot screen

DEBUG | ??:??:?? 7 [Screen] Show standard frames

DEBUG | ??:??:?? 7 [Screen] Show 0 module frames

DEBUG | ??:??:?? 7 [Screen] Total frame count: 103

DEBUG | ??:??:?? 7 [Screen] Added modules. numframes: 0

DEBUG | ??:??:?? 7 [Screen] Finished build frames. numframes: 3

DEBUG | ??:??:?? 8 [GPS] Probe for GPS at 115200

DEBUG | ??:??:?? 8 [GPS] Set Baud to 115200

DEBUG | ??:??:?? 9 [GPS] Trying $PDTINFO (Unicore Family)...

DEBUG | ??:??:?? 9 [GPS] Trying $PCAS06,1*1A (ATGM33xx Family)...

DEBUG | ??:??:?? 10 [GPS] Trying $PAIR021*39 (Airoha Family)...

DEBUG | ??:??:?? 11 [GPS] Trying $PQTMVERNO*58 (LC86)...

DEBUG | ??:??:?? 11 [GPS] Trying $PCAS06,0*1B (L76K)...

DEBUG | ??:??:?? 12 [GPS] Trying $PMTK605*31 (MTK Family)...

WARN | ??:??:?? 13 [GPS] No GNSS Module (baudrate 115200)

DEBUG | ??:??:?? 15 [GPS] Probe for GPS at 38400
[... 2347 lines cut ...]
DEBUG | ??:??:?? 2 Free heap : 30896 bytes

DEBUG | ??:??:?? 2 Free PSRAM : 0 bytes

DEBUG | ??:??:?? 2 [Screen] Screen: Started...

DEBUG | ??:??:?? 2 [Screen] haveGlyphs=1

DEBUG | ??:??:?? 2 [GPS] Probe for GPS at 9600

DEBUG | ??:??:?? 2 [GPS] Trying $PDTINFO (Unicore Family)...

DEBUG | ??:??:?? 3 [GPS] Trying $PCAS06,1*1A (ATGM33xx Family)...

DEBUG | ??:??:?? 3 [GPS] Trying $PAIR021*39 (Airoha Family)...

DEBUG | ??:??:?? 4 [GPS] Trying $PQTMVERNO*58 (LC86)...

DEBUG | ??:??:?? 5 [GPS] Trying $PCAS06,0*1B (L76K)...

DEBUG | ??:??:?? 5 [GPS] Trying $PMTK605*31 (MTK Family)...

Powered by [▲ Vercel](https://vercel.com/?utm_source=meshtastic&utm_campaign=oss) | Meshtastic® is a registered trademark of Meshtastic LLC. | [Legal Information](https://meshtastic.org/docs/legal).

Connected`

Basti77, 2025-04-17:
[image]
In the log, you can see that the device actually connects to the router and receives an IP address.
However, as soon as you try to connect to that IP, the device reproducibly crashes and reboots.

Pinging the device works, though.

For testing, I had an alpha firmware version installed, but the same issue occurs with the stable version as well.

Basti77, 2025-04-17:
When the log runs for a longer time, only this error keeps appearing.

again and again...

`DEBUG | 18:27:56 538 [ApiServer] PhoneAPI::close()

INFO | 18:27:56 538 [ApiServer] Incoming API connection

INFO | 18:27:56 538 [ApiServer] Incoming wifi connection

[538019][E][WiFiClient.cpp:67] fillBuffer(): Not enough memory to allocate buffer

[538035][E][WiFiClient.cpp:474] read(): fail on fd 52, errno: 11, "No more processes"

INFO | 18:27:56 538 [ServerAPI] Client dropped connection, suspend API service

INFO | 18:27:57 539 [ApiServer] Force close previous TCP connection`

d4rkmen, 2025-05-09:
Yes! I didnt notice this issue before and created a new one. Problem is not M5 Stack HW related, but the entire ESP32 platform. And the solution would be to refactor RAM usage.

Basti77, 2025-05-18:
I have now tried all alpha versions up to 2.6.8.
I can't adjust any improvements.