meshtastic/firmware #6607: [Board]: T5S3 aka T5 E-Paper S3 Pro

Xaositek, 2025-04-16:
### SOC

NRF52, ESP32

### Lora IC

SX1262

### Product Link

https://github.com/Xinyuan-LilyGO/T5S3-4.7-e-paper-PRO/tree/H752-01?tab=readme-ov-file

### Description

Adding this request based upon this product is shipping now from various sources.

pen25, 2025-04-20:
is there a bounty? id be willing to toss a fiver into the pot

agster27, 2025-04-29:
Would love to see this supported!

mr-tbot, 2025-06-06:
Agree!  Would love support for this device!

pen25, 2025-06-07:
im sure if someone sent a device to one of the devs they would get it supported. y9ou can also put up a bounty and ask others to donate
Phillip Ramirez

mr-tbot, 2025-06-08:
Happy to send my newer variant (with GPS) to the dev team if I eventually get it back!  Or happy to throw in to purchase one for the team and get it supported!

closed 2025-08-02 by github-actions[bot]

shiftingeden, 2026-03-25:
i can confirm as of last alpha, it still doesnt work. I dont think anyone is working on this problem.
However based on my tests, it seems to be related to  epdiy/src/board/epd_board_v7.c
from 
ESP_ERROR_CHECK(i2c_driver_install(EPDIY_I2C_PORT, I2C_MODE_MASTER, 0, 0, 0));
to
i2c_driver_install(EPDIY_I2C_PORT, I2C_MODE_MASTER, 0, 0, 0);

we need to make some changes in here
I would submit my changes in this file if anyone is interested

[epd_board_v7.c](https://github.com/user-attachments/files/26254556/epd_board_v7.c)

shiftingeden, 2026-04-05:
[image]
[image]

So far so good.
Bug in select region, you cannot confirm on device or connect phone .

giannoug, 2026-04-05:
**Update — boot reliability fixes & power management (sorry for the spam, I keep finding stuff to tinker with)**

Three more commits on the branch since the last update, all tested on H752-01 hardware.

---

**RTC chip fix**

The variant was defining `PCF85063_RTC` but the board carries a **PCF8563**. These are different chips with different register layouts — the wrong define caused the clock to silently fail on every boot. Fixed to `PCF8563_RTC` in `variant.h`.

---

**GT911 I2C address collision with SFA30 (ghost sensor)**

The GT911 touch controller powers up at I2C address `0x5D` by default (when the INT pin is left low/floating). `0x5D` is also the address of the SFA30 formaldehyde sensor — meaning the I2C bus scan run early in boot was finding the GT911 at `0x5D` and registering a false SFA30 detection. The telemetry module would then try to initialise the SFA30 at that address, get NACK'd by the GT911, and fail silently. There is no physical SFA30 on this board.

Per the GT911 datasheet §4.3 "Address Selection", the I2C address is latched during the RESET pulse based on the state of the INT pin:
- INT **HIGH** before RST deasserts → address **0x14**
- INT **LOW** before RST deasserts → address **0x5D**

Two fixes were applied:

1. `lateInitVariant()` already called `touch.begin(Wire, GT911_SLAVE_ADDRESS_H, ...)` to register at `0x14` — this was committed in the prior session. But the I2C scan runs *before* `lateInitVariant`, so the ghost detection was still occurring.

2. Added an `earlyInitVariant()` that drives INT HIGH, pulses RST, then releases INT — latching GT911 to `0x14` before the I2C scan runs. `lateInitVariant` still calls `touch.begin()` which repeats this sequence for full init; the double-reset is harmless and documented as such in `variant.cpp`.

---

**`variant_shutdown()` — clean deep sleep entry**

Added `variant_shutdown()` with two actions performed before the ESP32-S3 enters deep sleep:

- `touch.sleep()` — sends the GT911 sleep command, driving INT low and entering low-power standby
- `digitalWrite(BOARD_BL_EN, LOW)` — ensures the frontlight is off during sleep (a `LatchingBacklight` normally manages this, but an explicit pin-low is the safe fallback)

---

Branch: https://github.com/giannoug/meshtastic-firmware/tree/feature/t5s3-epaper-inkhud