meshtastic/firmware #6612: [Bug]: boot trying to update ntp time

fzellini, 2025-04-17:
### Category

WiFi

### Hardware

DIY

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4

### Description

After upgrading my DIY node based on Hydra, the WiFi stopped working and the device enters a boot loop after attempting to update the NTP time. I’ve attached the log output. Everything was working fine with the previous version. ( 2.5.xx)

### Relevant log output

```Shell
INFO  | ??:??:?? 6 Obtained IP address: 192.168.4.239
INFO  | ??:??:?? 6 Start network services
INFO  | ??:??:?? 6 mDNS Host: Meshtastic.local
INFO  | ??:??:?? 6 Start NTP time client
DEBUG | ??:??:?? 6 Init Web Server
INFO  | ??:??:?? 6 Start Secure Web Server
INFO  | ??:??:?? 6 Start Insecure Web Server
INFO  | ??:??:?? 6 Web Servers Ready! :-) 
INFO  | ??:??:?? 6 API server listen on TCP port 4403
[  6945][E][AsyncUDP.cpp:550] listen(): failed to start task
DEBUG | ??:??:?? 6 Failed to listen on UDP
DEBUG | ??:??:?? 7 [WifiConnect] Update NTP time from meshtastic.pool.ntp.org

abort() was called at PC 0x40085db9 on core 0

Backtrace: 0x40084139:0x3fffcf50 0x400976a1:0x3fffcf70 0x4009d4d5:0x3fffcf90 0x40085db9:0x3fffd010 0x40085de6:0x3fffd030 0x40085f5d:0x3fffd060 0x4015881a:0x3fffd080 0x40157a97:0x3fffd0a0 0x40157045:0x3fffd0c0 0x400934b5:0x3fffd0e0 0x40091f3e:0x3fffd100 0x40091fc8:0x3fffd120 0x400925d9:0x3fffd150 0x4022b92b:0x3fffd180 0x40223333:0x3fffd1a0 0x402234b1:0x3fffd4b0 0x4023d7c1:0x3fffd4e0 0x4017ffce:0x3fffd510 0x4018006b:0x3fffd540 0x40178c7d:0x3fffd5c0 0x4017d3c8:0x3fffd600

ELF file SHA256: da7527df1da18129
E (8360) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
E (820) esp_core_dump_flash: No core dump��ѥѥ���found!
E (820) esp_core_dump_flash: No core dump partition found!
[    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
@INFO  | ??:??:?? 0
```

benkyd, 2025-04-17:
More info needed about the specifics of the DIY setup

fzellini, 2025-04-17:
I am using this simple setup 

[image]

fzellini, 2025-04-18:
Other infos 
```
0x40084139: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
0x400976a1: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
0x4009d4d5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:46
0x40085db9: lock_init_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:75
0x40085de6: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:127
0x40085f5d: _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:167
0x401582b6: uart_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs_uart.c:208
0x40157533: console_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs_console.c:73
0x40156ae1: esp_vfs_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:431 (discriminator 4)
0x400934b5: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/stdio.c:94
0x40091f3e: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fflush.c:224
0x40091fc8: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fflush.c:278
0x400925d9: __sfvwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fvwrite.c:251
0x4022b3c7: __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:433
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:403
0x40222dcf: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1781 (discriminator 1)
0x40222f4d: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)
0x4023d25d: esp_log_writev at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/log/log.c:200
0x4017fa6a: __wrap_esp_log_writev at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp-insights/components/esp_diagnostics/src/esp_diagnostics_log_hook.c:452
0x4017fb07: __wrap_esp_log_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp-insights/components/esp_diagnostics/src/esp_diagnostics_log_hook.c:469
0x4017b08d: mdns_parse_packet at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c:3187 (discriminator 3)
0x4017cec5: _mdns_execute_action at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c:4518
 (inlined by) _mdns_service_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c:4644
```

fzellini, 2025-04-18:
Using
platform = https://github.com/platformio/platform-espressif32.git#master

I obtain this dump enabling wifi

```
~/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line -pfiaC -e .pio/build/hydra/firmware.elf 0x40084139:0x3ffd7ca0 0x400976a1:0x3ffd7cc0 0x4009d5d9:0x3ffd7ce0 0x4012cac9:0x3ffd7e10 0x400d39f3:0x3ffd7e30 0x4013217a:0x3ffd7e50 0x40134b3f:0x3ffd7ee0 0x40134bb6:0x3ffd7f00 0x4012ccf8:0x3ffd7f20 0x4011ae9e:0x3ffd7f50 0x4011a0ce:0x3ffd7f70 0x40131c00:0x3ffd8010 0x40131f62:0x3ffd8070 0x401377f6:0x3ffd8090 0x4014046e:0x3ffd80b0 0x40135df9:0x3ffd80d0 0x400e9521:0x3ffd8110
0x40084139: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
0x400976a1: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
0x4009d5d9: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
0x4012cac9: MemoryDynamic<_meshtastic_MeshPacket>::alloc(unsigned int) at ??:?
0x400d39f3: _ZN9AllocatorI22_meshtastic_MeshPacketE9allocCopyERKS0_j$constprop$1018 at ??:?
0x4013217a: Router::send(_meshtastic_MeshPacket*) at ??:?
0x40134b3f: FloodingRouter::perhapsRebroadcast(_meshtastic_MeshPacket const*) at :?
0x40134bb6: FloodingRouter::sniffReceived(_meshtastic_MeshPacket const*, _meshtastic_Routing const*) at ??:?
0x4012ccf8: ReliableRouter::sniffReceived(_meshtastic_MeshPacket const*, _meshtastic_Routing const*) at :?
0x4011ae9e: RoutingModule::handleReceivedProtobuf(_meshtastic_MeshPacket const&, _meshtastic_Routing*) at :?
0x4011a0ce: ProtobufModule<_meshtastic_Routing>::handleReceived(_meshtastic_MeshPacket const&) at :?
0x40131c00: Router::handleReceived(_meshtastic_MeshPacket*, RxSource) at :?
0x40131f62: Router::runOnce() at ??:?
0x401377f6: NextHopRouter::runOnce() at ??:?
0x4014046e: concurrency::OSThread::run() at ??:?
0x40135df9: loop() at ??:?
0x400e9521: loopTask(void*) at :?
```

fzellini, 2025-04-18:
compiled with debug info.... I believe there's not much memory available, and the Wi-Fi is consuming what little remains.
```
 ~/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line -pfiaC -e .pio/build/hydra/firmware.elf 0x400841c5:0x3ffd7f10 0x4009774d:0x3ffd7f30 0x4009d581:0x3ffd7f50 0x4021c027:0x3ffd7fd0 0x4021c06e:0x3ffd7ff0 0x4021c3b3:0x3ffd8010 0x4021c312:0x3ffd8030 0x4015c917:0x3ffd8050 0x401603a2:0x3ffd80a0 0x4016dc86:0x3ffd80c0 0x40116355:0x3ffd80e0 0x4015e9e2:0x3ffd8100 0x400da3e3:0x3ffd8120
0x400841c5: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
0x4009774d: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
0x4009d581: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:46
0x4021c027: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47
0x4021c06e: std::terminate() at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57
0x4021c3b3: __cxa_throw at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95
0x4021c312: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54
0x4015c917: APIServerPort<WiFiServerAPI, WiFiServer>::runOnce() at /home/fabrizio/meshtastic/firmware/src/mesh/api/ServerAPI.cpp:76
0x401603a2: non-virtual thunk to APIServerPort<WiFiServerAPI, WiFiServer>::runOnce() at ??:?
0x4016dc86: concurrency::OSThread::run() at /home/fabrizio/meshtastic/firmware/src/concurrency/OSThread.cpp:85
0x40116355: ThreadController::runOrDelay() at /home/fabrizio/meshtastic/firmware/.pio/libdeps/hydra/ArduinoThread/ThreadController.cpp:59
0x4015e9e2: loop() at /home/fabrizio/meshtastic/firmware/src/main.cpp:1318
0x400da3e3: loopTask(void*) at /home/fabrizio/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

fzellini, 2025-04-18:
checked out last git version ( 2.6.6.9da141aa8 ) and the problem disappear

closed 2025-04-18 by fzellini