meshtastic/firmware #6629: [Bug]: Direct Messages get stalled Queued for Sending

itxpress, 2025-04-19:
### Category

WiFi

### Hardware

Not Applicable

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4

### Description

After upgrading to firmware 2.64 (and 2.6.5) on nodes that have WiFI enabled, we've noticed a weird stalling issue on Messages. Normally with a Message, you see the empty cloud (Waiting to be acknowledged), then either the cloud w/slash (not delivered) or the check mark (Acknowledged) or person/checkmark (Delivery Confirmed)

Starting with 2.6.x on the nodes with WiFI, messgaes sent get stuck with a cloud w/up arrow 'Queued for sending' Nodes that don't have WiFI active never do this. The messages sometimes stay stuck like this forever. Other times they may stall for a minute or two, then all the queued messages get sent. After queued messgaes sned, you occasioanlly see normal send behavior (empty cloud waiting for acknowledgement), but then it'll revert. I'm seeing this with a Heltec v3, T-Beam, and T-Lora v2. Two of the nodes have MQTT active (differnet servers), but one does not and still has the issue.  They all see this behavior. We see it in any message sent - direct or to channels.  WiFI seems to be the commonality. Multiple other nodes without WiFI (Heltec, Rak 4631) send normally and never queue.

### Relevant log output

```Shell

```

fifieldt, 2025-04-19:
Thanks for your time debugging and writing up the report.

Would you be able to plug in one of your affected devices with USB, visit https://flasher.meshtastic.org  , tap Open Serial Monitor and then send through the logs that appear when this happens?

itxpress, 2025-04-23:
Finally got a chance to pull debug logs. I've attached them. First message I sent was "Sending another message" and I got the Cloud with up arrow queued icon. Second message was "Still stuck" same icon. Then after a bit, they sent. 3rd messages was 'Trying Again' and is sent normally (Empty cloud, then check in cloud) 4th message also sent normally. 

I noticed in the logs for the two messages that got stuck there was some kind of rate limit kicking in. For the 3rd message, those logs didn't happen. 

As I noted above. Sometimes a message stays queued and never sends. Other times it will eventually send after a random period of time.  Sometimes they send normally with no up arrow icon.

Note on this node, I do have S&F enabled, but on my Heltec v3, which also sees this issue, it's not enabled. The only common denominator seems to be WiFI. The TBeam has MQTT enable,d but the Heltec does not.  I just provisioned a TBeam Supreme with v2.6.4 I sent one message initially without issue. 15 minutes later tried to send another - it queued like the rest. Then it sent after a few minutes

[meshtastic-log-2025-04-23T01-44-35.944Z.log](https://github.com/user-attachments/files/19859588/meshtastic-log-2025-04-23T01-44-35.944Z.log)

oasis6212, 2025-05-22:
I can confirm I've seen this same behavior on two different heltec v3s running 2.6.4 both on wifi. Never experience it on blutooth.

marco-silva0000, 2025-05-25:
Just got this bug, have one in BT and one on the WiFi, the WiFi one can't DM the BT one

closed 2025-07-31 by github-actions[bot]