meshtastic/firmware #6636: [Feature Request]: Add Motion Telemetry Support for LIS3DH / RAK1904 3-Axis Accelerometer

neosmith20, 2025-04-20:
### Platform

NRF52

### Description

Hi Team,

I’d like to request official support for motion telemetry using the RAK1904 (LIS3DH) 3-axis accelerometer as part of Meshtastic’s telemetry framework.

It would be great to have a built-in module that:
 - Initializes the LIS3DH sensor
 - Captures X/Y/Z values
 - Sends that data over the mesh using the motion_metrics protobuf variant

This would be really useful for activity sensing, movement alerts, or device orientation tracking — and would complement the existing environmental telemetry features nicely.

Thanks for all the amazing work you’re doing!

garthvh, 2025-04-20:
The sensor is supported for several features but not as a sensor that sends data over the mesh. Sensors that send packets over the mesh need to be instant read.

neosmith20, 2025-04-20:
Thanks for the info! It’s great to hear the sensor is compatible for features like motion detection.

If it’s possible to implement it as an instant-read sensor for sending data over the mesh, that would be fantastic. It’d add a lot of value for users who want to utilize motion telemetry.

I’m curious — is there a way to make the LIS3DH sensor work as an instant-read within Meshtastic’s current framework?

Thanks again

garthvh, 2025-04-20:
You can use a number of different sensors with the detection sensor module.

closed 2025-06-12 by github-actions[bot]