meshtastic/firmware #6648: [Feature Request]: Support importing nodes into node db via a custom QR

RCGV1, 2025-04-21:
### Platform

Cross-Platform

### Description

Currently you need to discover a node to direct message it in application by importing a node (contact) via a qr code you can assure you are messaging the correct public key now that theres PKI. It could solve the public key mismatch issue where you had to delete the node and exchange instead, we can assume a node manually imported in is the real node. Also for ignoring a node you need them in your db so maybe you can share a node to ignore them in the future. Opens the possibility to a lot of features.

caveman99, 2025-04-22:
I have a slightly different idea. Are you familiar with the  swiss-made Threema Messenger? it's quite popular in Europe. They use a 'traffic lights' system for identity verification. 'red' means key mismatich, 'yellow' means 'i am using the key from the network handshake' and green means 'the key was verified by scanning the users QR code' - So the only people 'green' in your app are the ones you met in person to do the QR handshake. That's something we could implement 1:1 in Meshtastic, so you can still talk to over-the-air contacts but thy are displayed as the 'lesser' tier compared to the verified nodes.

ianmcorvidae, 2025-04-22:
I definitely like the idea of doing both of these. QR code scan (or whatever thing -- probably it's a User protobuf on some admin message, ultimately, so at least CLI probably doesn't literally scan a QR code) both bypasses the "don't update the key if you already have one" process, because you've verified in person, and sets a flag to indicate that higher level of trust for the green-light display. That flag can also behave like favorites do, in preventing those nodes from falling out of the nodedb, thus saving the effort of favoriting anyone you want to be sure to keep contact with.

RCGV1, 2025-06-15:
@thebentern Can you properly close this issue. Can't let the bot take the credit.

closed 2025-06-15 by garthvh