meshtastic/firmware #6649: [Bug]: SenseCAP Indicator new message

alexbadLV, 2025-04-22:
### Category

Hardware Compatibility

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [x] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4.b89355f

### Description

After receiving a message screen turn blue and stay until reboot, bot themes dark and Light.

[image]
[image]

[meshtastic-log-2025-04-22T04-02-29.927Z.log](https://github.com/user-attachments/files/19843744/meshtastic-log-2025-04-22T04-02-29.927Z.log)

### Relevant log output

```Shell

```

mverch67, 2025-04-22:
Does it also happen when the notification is set to "Banner only" ?

alexbadLV, 2025-04-22:
there are only 2 options first is banner and sound , second is silent
but with silent is the same

[image]
[image]
[image]

mverch67, 2025-04-22:
Go into BT programming mode, attach a phone with meshtastic app and disable **External Notification**. Then leave BT programming mode. That should do it. Do NOT enable the message alert switch in MUI.

alexbadLV, 2025-04-22:
Hi!
after this Go into BT programming mode, attach a phone with meshtastic app and disable External Notification.
I nave Sound only
receive the message no sound , no banner and NO BLUE!

[image]
[image]
[image]

mverch67, 2025-04-22:
Just tap on the bell icon (no long press) to switch to banner only.

alexbadLV, 2025-04-22:
nice at least banner! and no blue!

[image]

PR #6652 (2025-04-22): fix: SenseCAP Indicator: remove buzzer definition
fixes #6649 by removing the buzzer definition (which isn't connected to esp32 MCU)

## 🤝 Attestations
- [x] I have tested that my proposed changes behave as described.
- [x] I have tested that my proposed changes do not cause any obvious regressions on the following devices:
  - [ ] Heltec (Lora32) V3
  - [ ] LilyGo T-Deck 
  - [ ] LilyGo T-Beam
  - [ ] RAK WisBlock 4631
  - [ ] Seeed Studio T-1000E tracker card
  - [x] Seeed SenseCAP Indicator
  - [ ] Other (please specify below)
closed 2025-04-22 by thebentern