meshtastic/firmware #6655: [Feature Request]: GPIO trigger to wake from deep sleep in SENSOR role

redfast00, 2025-04-23:
### Platform

ESP32

### Description

(this might also be applicable for the NRF and RP2040 platform)

I would like to build a meshtastic letterbox notifier: when the lid is opened, a switch is pressed. This switch is connected to a GPIO pin of the Meshtastic device. It would then wake from deep sleep, send a message containing some telemetry (state of the switch, battery percentage), and go back to deep sleep. That way, it could last a very long time on battery.

Currently, you can set a power saving intervl for sensors, but not yet a GPIO trigger to wake it from deep sleep.

See https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/sleep_modes.html#api-reference-wakeup-source for how this could be implemented.

todd-herbert, 2025-04-23:
I know the user button wakes ESP32 devices from deep sleep. I haven't dug into its exact behavior with the sensor role, but I wonder if it might already be suitable for your application?

garthvh, 2025-04-24:
The detection sensor module exists for this purpose, the sensor role is for waking up and periodically sending out sensor readings.

GUVWAF, 2025-04-24:
Indeed the DetectionSensor module is suited for a letterbox detector, but it's based on polling and cannot yet be used for waking a device from sleep (interrupt-based).

redfast00, 2025-06-16:
Did a human decide 'not planned', or was this automatically done by github-actions?

rob-fi, 2026-02-02:
Also keen for this. I was just doing some research into putting a reed switch on my basement storage cage door with the idea of having a node in a deep sleep state until triggered.