meshtastic/firmware #6659: [Bug]: MUI Map does not support SD cards less than 2GB

thatSFguy, 2025-04-23:
### Category

Hardware Compatibility

### Hardware

T-Deck

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [x] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.5

### Description

I was running the initial "Nintendo Release Pre Release" for a long time.  At that time I had received a Map folder from another Meshtastic user.  I loaded it onto one of the small 128MB SD cards I had laying around and everything worked fine.  

I just upgraded to the latest Beta version and received the error of "Map Tiles not found!".  I then upgraded to the latest Alpha version thinking it was bug that was probably already fixed.  Since it still wasn't working I started checking the console and saw an error that it couldn't read the SDFS (sorry I didn't do a exact copy).  However, it recognized the SD just fine in terms of storage size.  

I then grabbed one of my 32GB SD drives and the map started working.  After expansive testing with multiple smaller drives, up to a 2GB SD, nothing under 2GB seems to be working even when formatted as FAT32 or ExtFAT.

It's not a big deal for me to use a bigger SD drive, but probably just should add a note to the docs:  https://meshtastic.org/docs/software/meshtastic-ui/

Thanks!

### Relevant log output

```Shell

```

b8b8, 2025-04-23:
Is this really an issue considering how cheap even 128gb sd cards are?

mverch67, 2025-04-24:
Since the first preview release 2.6.0 the SD driver has been replaced to support exFat (in 2.6.1). Seems this driver does not work with no longer purchasable SD cards.

closed 2025-04-24 by thebentern