meshtastic/firmware #6660: [Feature Request]: Dynamic light sleep support for ESP32 platform

m1nl, 2025-04-23:
### Platform

ESP32

### Description

**Platform**
ESP32

**Description**
Espressif provides advanced PM support in their ESP-IDF framework, which includes dynamic frequency scaling, dynamic light sleep (which maintains Bluetooth and WiFi connections) and other features. In the standard ESP32 Arduino PIO package, this feature is disabled and all `esp_pm_*` functions return ESP_ERR_NOT_SUPPORTED. Meshtastic has some initial implementation of DFS in `sleep.c`, but it won't work with the default ESP32 Arduino PIO package, and dynamic light-sleep is permanently disabled.

Current implementation of power-saving for ESP32 devices enables light-sleep periodically and is unsuitable for mobile devices as it disconnects device from the phone.

**Use Case**
I was able to implement dynamic light sleep feature PoC for Meshtastic firmware and I got about 30h of uptime on popular Heltec V3 platform with 1200mAh battery (H1 clone). This is HUGE improvement over standard battery life of about 8h. For most of the time, Bluetooth connection was active and I didn't have any issues communicating with other nodes. I consider this feature as stable and production ready as I'm using it with my other ESP32 projects (non-Arduino based).

**Current Workaround**
By replacing ESP-IDF in Arduino package, i was able to enable support for PM configuration and other required SDK configurations, which improve Bluetooth stability when light-sleeping. Relevant repositories are linked below. I also forked Meshtastic repository and created a dedicated branch with all changes, which enable dynamic light-sleep. Proposed changes don't break build for NRF devices (tested on heltec-t114) and also allows for legacy light-sleep functionality, when no support for PM is enabled in the ESP32 Arduino framework (i.e. default).

https://github.com/m1nl/arduino-esp32
https://github.com/m1nl/esp32-arduino-lib-builder
https://github.com/m1nl/meshtastic-firmware/tree/esp_light_sleep

**Benefits**
Meshtastic users would benefit from improved battery life on cheaper ESP32-based devices.

I wanted to open a discussion about this feature before submitting a PR, as it involves replacing the Arduino package. This change can only be applied to tested devices, so a staged roll-out is necessary - similar to the approach taken in the linked repository, which installs the modified package only for the heltec-v3 variant.

These dependencies should ideally be moved under the Meshtastic organization on GitHub, and we can add GitHub Actions to enable automated releases. Note that the default ESP-IDF version (v4.4) used by the mainline package in PlatformIO (provided by Espressif) is unsupported; the same version is currently used in the modified package. An upgrade can be considered at a later stage.

Once there's a decision on adopting the customized Arduino package, I’ll proceed with creating a PR to integrate the changes into the Meshtastic firmware.

m1nl, 2025-04-23:
[image]

EternityForest, 2025-05-13:
There is a way to use this automatic light sleep in a slightly simpler way, with the dual IDF/Arduino mode, just setting both Arduino and ESP-IDF on PlatformIO's framework setting.

I'm using it in one of my projects:

https://github.com/EternityForest/ArduinoCogs/tree/main/examples/espidf-arduino-platformio

m1nl, 2025-05-14:
ok thanks, I will give it a try soon

EternityForest, 2025-05-17:
Sigh! It appears Meshtastic uses a few features like src_filter, and some of the predefined board macros, which are not fully trivial to get working with the dual IDF/Arduino approach...

nbun, 2025-05-21:
I can confirm the improvements. My Heltec V3 went from roughly 20 hours on a 2000 mAh cell with active WiFi to about 48 hours.