meshtastic/firmware #6662: [Board]: roamer

janveeh, 2025-04-24:
### SOC

NRF52

### Lora IC

SX1262

### Product Link

https://www.tindie.com/products/embeddedarts/roamer-meshtastic-wireless-lora-mesh-comms/

### Description

**Handheld "rugged" device based on the RAK4630. Functionality tested and PR ready.**

The device has the following features:
- RAK4630 module
- SSD1306 OLED screen
- Rotary Encoder
- uBlox M10Q GPS module
- 2000mAh battery for up to 1 week of continous use
- 3D-printed Nylon case

Open issues:
- GPS power consumption could be optimised by using M10 ultra low power features
- When using the encoder button for the canned message module, the device can't be switched on or off anymore

RCGV1, 2025-04-24:
I don’t think your case design is compliant with the Meshtastic. trademark guidelines.

https://meshtastic.org/docs/legal/licensing-and-trademark/

janveeh, 2025-04-24:
Thanks for the hint, case design will be revised.

fifieldt, 2025-06-08:
Hi @janveeh , thanks for following the trademark guidelines. Please go ahead and make a PR for the board and variant :) We're here to help