meshtastic/firmware #6666: [Bug]: HTTPS issues on RAK11200

db3000, 2025-04-25:
### Category

WiFi

### Hardware

Rak11200

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4.b89355f

### Description

HTTPS access is very unreliable on my new RAK11200 device. Connections via HTTP seem to go through fine, but when connecting via HTTPS (as required by client.meshtastic.org) the device is largely unreachable. With a fresh boot a few connections will go through, but then all future connections will be dropped.

My test procedure:
1) Full erase and flash using flasher.meshtastic.org
2) Connect via serial from client.meshtastic.org
3) Turn on Wifi
4) Turn off Bluetooth
5a) Go to https://meshtastic.local/ in Chrome and click through the certificate warning
5b) Try to access the device via HTTPS using client.meshtastic.org
6) Run curl -k https://meshtastic.local from the command line

I have tried the above with firmware 2.5.20, 2.6.4 and 2.6.5.

Connecting via HTTP:
```
$ curl http://meshtastic.local
Web server is running.<br><br>The content you are looking for can't be found. Please see: <a href=https://meshtastic.org/docs/software/web-client/>FAQ</a>.<br><br><a href=/admin>admin</a>
```

Connecting via HTTPS:
```
$ curl -v -k https://meshtastic.local
* Host meshtastic.local:443 was resolved.
* IPv6: (none)
* IPv4: x.x.x.x
*   Trying x.x.x.x:443...
* Connected to meshtastic.local (x.x.x.x) port 443
* ALPN: curl offers h2,http/1.1
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* Recv failure: Connection reset by peer
* OpenSSL SSL_connect: Connection reset by peer in connection to meshtastic.local:443
* Closing connection
curl: (35) Recv failure: Connection reset by peer
```

On the serial console when a connection is dropped:
```
[HTTPS:I] New connection. Socket FID=52
[HTTPS:E] SSL_new failed. Aborting handshake. FID=52
[HTTPS:I] Connection closed. Socket FID=52
```

Or sometimes I get this error on the serial console:
```
[HTTPS:I] New connection. Socket FID=52
[HTTPS:E] SSL_accept failed. Aborting handshake. FID=52
[HTTPS:I] Connection closed. Socket FID=52
```

With a fresh boot I can get client.meshtastic.org to load sometimes, but the board quickly either reverts to rejecting connections or crashes after a few HTTPS requests. Crash log:
```
DEBUG | 02:39:37 36 [WebServer] webAPI handleAPIv1FromRadio
[HTTPS:E] An receive error occured, FID=52
[HTTPS:I] Connection closed. Socket FID=52
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4023338b  PS      : 0x00060730  A0      : 0x800f77a0  A1      : 0x3ffd7de0
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000003f  A5      : 0x3ffd7da0
A6      : 0x3ffd7da0  A7      : 0x00000001  A8      : 0x800f7a81  A9      : 0x3ffd7dc0
A10     : 0x3fffeba8  A11     : 0x00000000  A12     : 0x00000020  A13     : 0xffffffff
A14     : 0x3ffd7d60  A15     : 0x00000008  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000018  LBEG    : 0x40093dfc  LEND    : 0x40093e12  LCOUNT  : 0xffffffff

Backtrace: 0x40233388:0x3ffd7de0 0x400f779d:0x3ffd7e00 0x400f78c3:0x3ffd7e20 0x400fde12:0x3ffd7e40 0x400fc795:0x3ffd7e60 0x400fcff5:0x3ffd8050 0x40126781:0x3ffd8080 0x40140742:0x3ffd80a0 0x40135edd:0x3ffd80c0 0x400e9581:0x3ffd8100
```
after which the board reboots.

Please let me know what other debugging steps I can take.

### Relevant log output

```Shell

```

todd-herbert, 2025-04-25:
I don't have any particular insight myself here, sorry, but I did take a few moments to decode that error message to give anyone else a clue:

```
0x4023338b: mbedtls_mpi_get_bit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c:291

0x40233388: mbedtls_mpi_get_bit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c:291
```

db3000, 2025-04-30:
I was hoping that 2.6.6 which includes PR https://github.com/meshtastic/esp32_https_server/pull/4 would fix this. But still same issue on 2.6.6: 
```
[HTTPS:I] New connection. Socket FID=52
[HTTPS:E] SSL handshake failed. Aborting handshake. FID=52
[HTTPS:I] Connection closed. Socket FID=52
```

And occasional crash log for 2.6.6:
```
DEBUG | ??:??:?? 81 [WebServer] webAPI handleAPIv1FromRadio
[HTTPS:E] An receive error occured, FID=52
[HTTPS:I] Connection closed. Socket FID=52
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0xc2a26a9b  PS      : 0x00060b30  A0      : 0x800fd367  A1      : 0x3ffd7df0
A2      : 0x3ffed7bc  A3      : 0x3fff1758  A4      : 0x00000003  A5      : 0x3fffd37c
A6      : 0x3fffd354  A7      : 0x3fffd33c  A8      : 0x800f8e45  A9      : 0x3fffd010
A10     : 0x3f40a918  A11     : 0x3fffd33c  A12     : 0x00000002  A13     : 0x00000001
A14     : 0x3fff2748  A15     : 0x3ffd7d10  SAR     : 0x0000001a  EXCCAUSE: 0x00000014
EXCVADDR: 0xc2a26a98  LBEG    : 0x40093e65  LEND    : 0x40093e87  LCOUNT  : 0xffffffff

Backtrace: 0x42a26a98:0x3ffd7df0 0x400fd364:0x3ffd7e10 0x400fc77a:0x3ffd7e30 0x400fcf65:0x3ffd8020 0x4012690d:0x3ffd8050 0x4014096a:0x3ffd8070 0x40136105:0x3ffd8090 0x400e94b5:0x3ffd80d0

ELF file SHA256: af898a9bb56f8e48
```

db3000, 2025-04-30:
Added some debugging, looks like I get a lot of `MBEDTLS_ERR_SSL_ALLOC_FAILED` errors from `mbedtls_ssl_setup` calls. So possibly this is a memory issue?

db3000, 2025-05-05:
I have a PR ready for the crash and the timeout issue, but I'm wondering if there is any interest in just dropping the esp32_https_server library and replacing it with something else? Given that there are longstanding bugs in there and it is basically unmaintained, plus changing the existing code to preallocate memory for the buffers would not be straightforward, it seems like maybe there is a newer maintained ESP32-compatible HTTP server that could be used instead?

thebentern, 2025-05-05:
> I have a PR ready for the crash and the timeout issue, but I'm wondering if there is any interest in just dropping the esp32_https_server library and replacing it with something else? Given that [...]

Absolutely! Do you have an alternative in mind?

db3000, 2025-05-05:
Not specifically, but can try out some other options this week.

todd-herbert, 2025-05-05:
Totally unrelated, but that library has been causing headaches for the web client because it doesn't traverse sub-directories. Changing it out could be a two birds with one stone situation!

db3000, 2025-05-13:
Have been playing around a bit with https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/esp_https_server.html, this is definitely maintained and likely has fewer bugs but its a big job to convert everything. Also, at the core this option would still be using mbedtls which allocates 16k buffers dynamically by default, potentially running into the same heap fragmentation.

Still looking around, are there any other options? Everything else I find for ESP32 HTTPS support is either archived/unmaintained or seems largely unused.

For now, maybe I'll send in my PRs to fix the smaller issues with esp32_https_server.

wojo, 2025-05-21:
Heltec V3 units crash with HTTPS as well, HTTP is fine. Can upgrade on a alpha version to test.