meshtastic/firmware #6672: [Bug]: No signal on SenseCap Indicator

alexbadLV, 2025-04-25:
### Category

Hardware Compatibility, Other

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [x] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4.b89355f Beta

### Description

I got no signal after this
and node online came to 1 its mine!

[image]

[meshtastic-log-2025-04-25T18-16-25.581Z.log](https://github.com/user-attachments/files/19914221/meshtastic-log-2025-04-25T18-16-25.581Z.log)

### Relevant log output

```Shell
INFO | 19:25:35 132227 [Router] Received Admin from=0x0, id=0xfd59cc43, portnum=6, payloadlen=3

INFO | 19:25:35 132227 [Router] Handle admin payload 16

INFO | 19:25:35 132227 [Router] Client got device connection status

DEBUG | 19:25:35 132227 [Router] Set admin key to 34 3b 07 a1 ab 9e 8e a5

DEBUG | 19:25:35 132227 [Router] Partially randomized packet id 2436675652

INFO | 19:25:35 132227 [Router] Asked module 'Admin' to send a response

DEBUG | 19:25:35 132227 [Router] Module 'neighborinfo' wantsPacket=1

INFO | 19:25:35 132227 [Router] Received neighborinfo from=0x0, id=0xfd59cc43, portnum=6, payloadlen=3

DEBUG | 19:25:35 132227 [Router] Module 'neighborinfo' considered

DEBUG | 1900 fr=0x00000000 to=0xda7522d8, WantAck=1, HopLim=0 Ch=0x0 Portnum=6 WANTRESP)

DEBUG | 21:14:15 138747 [PacketAPI] Partially randomized packet id 3366513010

DEBUG | 21:14:15 138747 [PacketAPI] Enqueued local (id=0xc8a8e972 fr=0x00000000 to=0xda7522d8, WantAck=1, HopLim=0 Ch=0x0 Portnum=6 WANTRESP rxtime=1745604855)

DEBUG | 21:14:15 138747 [Router] Add packet record fr=0x0, id=0xc8a8e972

DEBUG | 21:14:15 138747 [Router] handleReceived(REMOTE) (id=0xc8a8e972 fr=0x00000000 to=0xda7522d8, WantAck=1, HopLim=0 Ch=0x0 Portnum=6 WANTRESP rxtime=1745604855)
DEBUG | 21:14:15 138747 [Router] Module 'Admin' wantsPacket=1

INFO | 21:14:15 138747 [Router] Received Admin from=0x0, id=0xc8a8e972, portnum=6, payloadlen=3

INFO | 21:14:15 138747 [Router] Handle admin payload 16

INFO | 21:14:15 138747 [Router] Client got device connection status

DEBUG | 21:14:15 138747 [Router] Set admin key to 93 45 a2 2e 63 00 ae 69
```

mverch67, 2025-04-25:
This is not a bug. You lost connection to all other nodes, so it goes to state "no signal".

alexbadLV, 2025-04-26:
Ok. but it stay in this state forever, I need to manualy restart the device!

alexbadLV, 2025-04-26:
Hi Manuel!
Ok. but it stays in this state forever, I need to manually restart the
device!
And I usually have about 20 active nodes around me!
and it shows me them after restart!

mverch67, 2025-04-26:
I suggest you update the description in your bug report. It wasn't clear what you meant.

alexbadLV, 2025-04-26:
After restart an d few minutes I got back lots of Online nodes.

[image]

[meshtastic-log-2025-04-26T10-22-24.026Z.log](https://github.com/user-attachments/files/19921729/meshtastic-log-2025-04-26T10-22-24.026Z.log)

alexbadLV, 2025-04-26:
last update 28 minutes uptime and I got almost all nodes online
it just cant be that in some random time they all went offline!

[image]

alexbadLV, 2025-05-04:
I dont know is it a coincidence or what, got a new antenna, now always positive SNR and no more No signal!
Will test longer but now 3 days online!

[image]

fifieldt, 2025-05-04:
Q) when did you buy your device? I heard that some very early models had an issue but don't know any details...

alexbadLV, 2025-05-04:
Already return my first purchase and got new version about a month ago!

wehooper4, 2025-06-04:
I want to report I'm having the same issue with one of these. Works for between 2 and 24 hours, then stops receiving/transmitting packets. 

I was purchased a month ago, not the old batch that had issues (I have two of those as well, which oddly DON'T have this issue, but only will send big packets on faster spread factors)

mighty-takahe, 2025-06-06:
I've got the same issue with MQTT and wifi on. I've not tested MQTT and wifi off yet.

I've only just contacted Seeed Studio support, and they've yet to examine my logs, but they've suggested that the issue may be that I have a lot of nodes around me (usually around 90 out of 170), and if they're sending messages to each other it may cause a memory overflow.

misiekkk, 2025-08-12:
Hi all, I have this some issue with Wifi enabled, GPS enabled, about 250 nodes in DB.
I will make step test:
- tomorrow with disabled WiFi
- next - disabled Wifi and GPS
I have 3 another (different) nodes which work 2-3 weeks without any issues or freezing.