meshtastic/firmware #6674: [Bug]: Firmware 2.6.2 continually loses BT connectivity and is difficult to enter DFU mode

Geordon, 2025-04-26:
### Category

BLE

### Hardware

Seeed Card Tracker T1000-E

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.2.31c0e8f

### Description

Just received my new Seeed T1000-e and went to update the firmware to the above. After a few minutes, the unit stopped responding to Bluetooth.  In order for me to re-connect to the unit, I had tp monkey around to get it to restart, only to have to create a new pairing connection.

Falling back to version 2.5.20.4c97351 has made everything work as expected.

### Relevant log output

```Shell

```

Xaositek, 2025-06-09:
I would highly recommend going into the Flasher, utilize the Erase Flash UF2 file and then reload Meshtastic and reconfigure. Make sure you back up your keys before you reset the device

Xaositek, 2025-07-03:
Closing this - many fixes for Bluetooth in Meshtastic 2.7, please wipe/reload with newer firmware and open a new issue if still occurring.

closed 2025-07-03 by Xaositek

kozzae, 2025-08-21:
Hopefully someone will know ? So on my RF50 xio seeed I can't get it to come up with DFU so that it programs (using chrome on win11 ) I also have a 1000e tracker think I ordered the wrong one mines is a lorawan can it be re flashed to work Meshtastic ??