meshtastic/firmware #6675: [Feature Request]: Enhance telemetry algorithm

b8b8, 2025-04-27:
### Platform

Cross-Platform

### Description

Currently the telemetry is rate limited in two ways.
1. Hard limit for clients at 25% channel utilization
2. Scaling broadcast interval when online node count over 40 (and faster telemetry when under 10) nodes.
Keep these as they are.

Larger meshes see a lot of telemetry data that congests the mesh when node counts are over about 100 nodes. 
People are buying multiple meshtastic devices and even though they may have one "main" node, they have several more in their house turned on for telemetry, testing, etc. They care about this telemetry data.

My idea is to update the broadcast interval to a 2 hour hard limit for standard telemetry (currently 1 hour default, 30min minimum) that follows the user set hop count, and then add zero hopped telemetry at 1/4 the rate (30min) of the set broadcast interval.

Similar situation with position data but 6 hour broadcast minimum with hop limit, 1/4 rate for zero hopped position, + smart position settings.

This allows people with multiple nodes and nodes within their neighborhood to get more frequent telemetry (and position) updates while not carrying that forward into the rest of the mesh, especially through mountain routers.

This still allows all messages out to the rest of the mesh as well as periodic telemetry on the rest of the mesh nodes at a controllable rate that is less frequent to the less relevant nodes.

jensoliver, 2025-05-04:
I do not agree on limiting position. We are using the nodesin rescue operations and are in need of positions of the team members every 5 minutes

b8b8, 2025-05-04:
> I do not agree on limiting position. We are using the nodesin rescue operations and are in need of positions of the team members every 5 minutes

This wouldn't and isn't supposed to affect you. Its for larger mesh networks that are public and doesn't affect smart position anyway. Smart position set to 10m 2min for example would be fine. I use meshtastic for hiking and similar scenario to you. The point is to not mess with any of our preferred uses, just dial it back automatically when in a city of 100 or more nodes where lots of people have a node in a stationary house.

b8b8, 2025-06-08:
> I do not agree on limiting position. We are using the nodesin rescue operations and are in need of positions of the team members every 5 minutes

Out of curiosity are your nodes in tracker role when used in rescue operations?

wlockwood, 2025-09-16:
Being able to turn off telemetry entirely (which doesn't seem to work currently) for nodes where the data is wrong or not meaningful would be great. In some cases, we need it so that nodes can discover each other, but that's really only relevant in the first tiny bits of a new mesh.
Being able to send position and telemetry to different channels or even target nodes would also give us a lot of flexibility.