meshtastic/firmware #6676: firmware support Chinese input when sending message

zhjygit, 2025-04-27:
like above.
When a node sends message to another node, We can input message in Chinese.

todd-herbert, 2025-04-27:
By coincidence, I tried today with NRF52, but it didn't work because there was not enough memory. I remember that someone made it work a few months ago, but it needed ESP32 and TFT.

<https://github.com/meshtastic/firmware/issues/5895>
It seems promising, even if it only works for a few devices. Hopefully someone can develop this more!

zhjygit, 2025-04-27:
https://docs.lvgl.io/9.2/others/ime_pinyin.html#example

zhjygit, 2025-04-27:
> https://docs.lvgl.io/9.2/others/ime_pinyin.html#example

with this link,maybe we can add the pinyin input to the t-deck.

zhjygit, 2025-04-27:
I just want to send message to others,not display. how can i use "lv_ime_pinyin" api in the main meshtastic/firmware project

todd-herbert, 2025-04-27:
Oh sorry, I misunderstood. I never worked on the T-Deck, so I don't know how, sorry, but pinyin input would be a good feature!

closed 2025-06-20 by github-actions[bot]