meshtastic/firmware #6688: RXEN pin for XIAO NRF kit appears to be wrong

NomDeTom, 2025-04-29:
@Dylanliacc should the RXEN pin on the (new) XIAO kit be D5 rather than D4? This would currently seem to conflict with the CS pin for the board.

https://github.com/meshtastic/firmware/blob/635de2d2296d463fb61fe5a650bd06b94a6559ee/variants/seeed_xiao_nrf52840_kit/variant.h#L117

https://www.seeedstudio.com/XIAO-nRF52840-Wio-SX1262-Kit-for-Meshtastic-p-6400.html

https://www.seeedstudio.com/Seeed-XIAO-BLE-nRF52840-p-5201.html

madeofstown, 2025-04-30:
I did point this out when the variant was updated recently.

mverch67, 2025-05-17:
Looks like it's fixed in 2.6.8 (#6717)

mverch67, 2025-05-17:
Looks like it's fixed in 2.6.8 (https://github.com/meshtastic/firmware/pull/6717)

closed 2025-05-17 by mverch67