meshtastic/firmware #6698: [Feature Request]: Multi node / multi preset teaming

humbertovnavarro, 2025-04-30:
### Platform

other

### Description

Add the ability to chain together multiple devices/and or radios to work in parallel across serial or spi. Perhaps some sort of auto negotiation and shared lookup table combined with custom presets that define the behavior of the nodes. I.E. a "bond" preset that allows message forwarding between different presets. This feature would alleviate the fracturing of communities across presets as node density increases. I'm not sure what the ramifications of this would be but I think it's worthy of discussion. Also - lots of nodes on a single pcb looks cool.

b8b8, 2025-04-30:
https://github.com/meshtastic/firmware/issues/4760#issue-2532590911

garthvh, 2025-04-30:
Duplicate of https://github.com/meshtastic/firmware/issues/4760#issue-2532590911

closed 2025-04-30 by garthvh