meshtastic/firmware #6704: [Feature Request]: Pull Neighbor list remote

HougeDK, 2025-04-30:
### Platform

NRF52

### Description

Hello.

Would it be possible to make a way to pull a neighbor list from a remote node.
I would like to see what my remote nodes can pick up on hop 0, but i don´t know any way to do that.
I tried the neighbor via lora option, but i don´t see any way to view the information.

Cheers
Brian.

DaneEvans, 2025-06-16:
I'm looking into this one as it aligns with my plans.

HougeDK, 2025-06-17:
Excellent, im happy to hear that.

DaneEvans, 2025-08-11:
Still planning on this one.

DaneEvans, 2025-10-22:
Not stale, just busy.  
I have a WIP on it. 

Yeah, neighbour info does it, but not in a way that is useful for many of our use cases.

DaneEvans, 2025-11-03:
MR in, 
Sorry HougeDK, no RSSI. I was able to tack it onto the existing setup with minimal changes, so that should help it get merged quicker.