meshtastic/firmware #6725: [Bug]: nRF52840 ProMicro Nano v2 + SH1107 incorrect image display

Samekk, 2025-05-02:
### Category

Hardware Compatibility

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.6.54c1423 Alpha

### Description

After connecting a 1.5" 128x128 OLED (GME128128-01-IIC) display module and selecting SH1107, the image displayed on the screen is not cut off/displayed on part of the screen.

[image]

### Relevant log output

```Shell

```

Edekoo10, 2025-05-02:
I have the same problem :(
PLS fix it :)

markbirss, 2025-05-07:
you need to modify variant.h and define specially the correct display type

`
#define HAS_SCREEN 1
#define USE_SH1107
`

closed 2025-06-29 by github-actions[bot]