meshtastic/firmware #6730: [Board]: GAT562 Mesh Trial Tracker

hayschan, 2025-05-04:
### SOC

NRF52

### Lora IC

SX1262

### Product Link

https://meshcn.net/GAT-IOT-handheld-review/

### Description

The GAT562 Mesh Trial Tracker is a ready-to-use Meshtastic handheld device that eliminates the need for assembly or soldering. It integrates a variety of essential features into one compact device, including a battery, screen, GPS, and antenna, which are typically required to be assembled separately in other products. This device is designed to make the Meshtastic experience more accessible to users who may not have prior hardware assembly experience.

[image]

Powered by the efficient nRF52 chipset, the GAT562 is built for low power consumption, making it an excellent choice for long-term usage in the field. It supports LoRa communication across the 470 MHz and 915 MHz frequency bands, ensuring compatibility with both domestic and international LoRa networks. The device also includes a 2500mAh battery that offers a promising battery life, which is expected to last up to 48 hours with full functionality, although practical use may yield even longer runtimes.

[image]

The device comes pre-loaded with the latest Meshtastic firmware, ensuring that it is ready to use out of the box without the need for additional configuration or hardware. The 1.3-inch OLED screen offers clear visibility, even in bright outdoor conditions, thanks to the self-illuminating properties of OLED technology.

hayschan, 2025-05-04:
I am writing the code to complete this board addition. I'll be creating a pull request in a few days.

NomDeTom, 2025-05-04:
@hayschan Is this your design? It looks really cool, and I'm keen to see what parts you've used and the customisation you've made.

There is one point worth noting: in some of the promotional photos, the case design shows the Meshtastic M logo, without any of the "community" modifications.

It is worth reading the [requirements for using the logo here](https://meshtastic.org/docs/legal/licensing-and-trademark/) and potentially making a few small changes before you take it any further.

Anyway, it looks awesome, and I'm sure we'll see interesting stuff soon.

jp-bennett, 2025-05-04:
@hayschan You can drop us an email at contact@meshtastic.com for help with getting this compliant with the trademark rules, upstreamed into the project, and listed on the flasher.

Thanks!

hayschan, 2025-05-05:
> There is one point worth noting: in some of the promotional photos, the case design shows the Meshtastic M logo, without any of the "community" modifications.
> It is worth reading the [requirements [...]

Thanks for the reminder. Yep, I actually realize the logo issue yesterday when I am looking at another Board Addition issue https://github.com/meshtastic/firmware/issues/6662. I saw comment regarding the trademark on that post as well. 

I'll definitely remove the Meshtastic logo from the case to comply with the logo trademark policy. @NomDeTom

hayschan, 2025-05-05:
> [@hayschan](https://github.com/hayschan) You can drop us an email at [contact@meshtastic.com](mailto:contact@meshtastic.com) for help with getting this compliant with the trademark rules, upstreamed into the project, and listed on the flasher.
> 
> Thanks!

Thanks. Didier Barbas has contacted me. Interestingly, I have talked with Didier Barbas previously already, not related to this issue though. He is also a member of MeshCN community. He's also in the group chat of the community.

fifieldt, 2025-05-05:
Thanks for being a great open source collaborator and supporter of Meshtastic :)

hayschan, 2025-05-05:
After reminding GAT-IOT, they have now removed the Meshtastic logo from the case design. @NomDeTom 

[image]

MkMunich, 2025-05-06:
Hi, should GAT562 also support European Union 868MHz frequency or is the hardware & antenna restricted to the given frequencies?

hayschan, 2025-05-08:
> Hi, should GAT562 also support European Union 868MHz frequency or is the hardware & antenna restricted to the given frequencies?

It support 868 MHz as well.

[image]

hayschan, 2025-05-13:
Just created a pull request https://github.com/meshtastic/firmware/pull/6805 for this model. Am I contributing things correctly @NomDeTom ?

fifieldt, 2025-07-07:
This board has turned out to be quite popular, we should finish this work :)

hayschan, 2025-07-10:
> This board has turned out to be quite popular, we should finish this work :)

I believe the firmware side is done by @csrutil on his PR https://github.com/meshtastic/firmware/pull/6984 already. 

However, the web flasher part PR is still pending for review https://github.com/meshtastic/web-flasher/pull/193 .

closed 2026-09-03 by garthvh