meshtastic/firmware #6731: [Feature Request]: Using Meshtastic devices as a location provider

k8ieone, 2025-05-04:
### Platform

NRF52, ESP32

### Description

I'm not sure if Meshtastic is the right place for this, but would it be possible to use a Meshtastic device with an installed GPS module as a location source for devices without a GPS (laptops, etc.)?

Concrete use-case: I have a Linux-powered phone (effectively a laptop). The integrated GPS receiver sucks and I haven't been able to get a location fix for hours. Since the GPS module on Meshtastic devices is usually pretty good, it seems like it could be a great solution.

Another use-case: Meshtastic devices are pretty cheap. Someone looking to build a cheap DIY navigation system could buy a cheap Android tablet and a GPS-enabled Meshtastic device, connect the two together and voilà! They'd have a decent working navigation system.

It seems to me like external GPS USB or Bluetooth receivers are becoming a rarity. Those that I was able to find are really expensive for what they are.

I think the technical details should be fairly simple. Output NMEA 0183 over USB serial or a Bluetooth [SPP](https://www.bluetooth.com/specifications/specs/serial-port-profile-1-2/). [gpsd](https://gpsd.gitlab.io/gpsd/bt.html) should then be able to work with it.

*Disclaimer: I have no idea what I'm talking about. I'm not familiar with how any of those external GPS devices work. The last time I saw one of those things was in ~2007.*

My biggest worry is that this is too far outside the scope of Meshtastic and would be more suited for a custom firmware that only works as a GPS provider.

closed 2025-05-04 by thebentern

k8ieone, 2025-05-04:
@thebentern Is it possible to disable everything else from logging to the serial console?

I'm getting stuff mixed in with the NMEA output and I think `gps-share` is getting confused by it. 

```
INFO  | 14:37:56 426 [Serial] BLE disconnect
*RAW NMEA LINE*
INFO  | 14:37:57 427 [GPS] GPS power state move from ACTIVE to IDLE
INFO  | 14:37:57 427 [GPS] updatePosition LOCAL *omitted*
INFO  | 14:37:57 427 BLE disconnect
*RAW NMEA LINE*
*RAW NMEA LINE*
INFO  | 14:38:02 432 [GPS] GPS power state move from IDLE to ACTIVE
INFO  | 14:38:02 432 BLE disconnect
INFO  | 14:38:02 432 [GPS] GPS power state move from ACTIVE to IDLE
INFO  | 14:38:02 432 [GPS] updatePosition LOCAL *omitted*
*RAW NMEA LINE*
*RAW NMEA LINE*
*RAW NMEA LINE*
INFO  | 14:38:07 437 [GPS] GPS power state move from IDLE to ACTIVE
```

k8ieone, 2025-05-04:
Got it, it was under security - https://meshtastic.org/docs/configuration/radio/security/#serial-console

k8ieone, 2025-05-04:
I was able to successfully use a Heltec Wireless Tracker connected over USB to provide location info to a laptop which is exactly what I wanted.

Details:

- The laptop runs [`gnss-share`](https://gitlab.postmarketos.org/postmarketOS/gnss-share)
- `gnss-share` is pointed at the serial device and provides a socket at `/var/run/gnss-share.sock`
- AFAIK all modern apps use Geoclue for interacting with location stuff
- Geoclue is pointed at the `gnss-share` socket and provides location info to all apps that need it (tested with [Pure Maps](https://github.com/rinigus/pure-maps))

[`gnss-share.conf`](https://gitlab.postmarketos.org/postmarketOS/gnss-share/-/blob/master/gnss-share.conf):

```
socket="/var/run/gnss-share.sock"
group="geoclue"
device_driver="stm_serial"
device_path="/dev/ttyACM0"
# Has to match serial baud rate configured in Meshtastic
device_baud_rate=9600
```

[`geoclue.conf`](https://man.archlinux.org/man/extra/geoclue/geoclue.5.en):

```
[network-nmea]
enable=true
nmea-socket=/run/gnss-share.sock
```

k8ieone, 2025-05-04:
Doing the same over Bluetooth would be taking it to a next level, but I think that would require changes to the Meshtastic firmware. I'm not sure what the support for using external GPS modules is like on Android and iOS these days.

I do know that it would be possible on Linux if the NMEA sentences were sent over [SPP](https://www.bluetooth.com/specifications/specs/serial-port-profile-1-2/). One would just need to run [`rfcomm`](https://github.com/zeenix/gps-share?tab=readme-ov-file#supported-devices) (or some modern alternative to that since it is deprecated) to create a local serial device. From that point the procedure should be exactly the same.

thebentern, 2025-05-04:
> Doing the same over Bluetooth would be taking it to a next level, but I think that would require changes to the Meshtastic firmware. I'm not sure what the support for using external GPS modules is [...]

Sadly most of the specs I've seen for location sharing has been for classic bluetooth rather than BLE, but if you find something different let me know. It's something I've thought about before as a lot of the commercial products are honestly overpriced when compared with even the most expensive meshtastic compatible hardware