meshtastic/firmware #6738: [Bug]: SEED ESP32S3 & WIO SX1262, "No Interface" Error When Setting GPS RX/TX Pins in Meshtastic App

Genei180, 2025-05-05:
### Category

Hardware Compatibility

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4.b89355f (Stabel Beta)

### Description

Summary:
After flashing two devices with the firmware, they can successfully exchange messages. However, when attempting to set up a GPS module via the Meshtastic app and Bluetooth, critical issues occur.

Steps to Reproduce:
- Connect to the device via Bluetooth using the Meshtastic app.
- Open the Position Settings.
- Set RX to Pin 6 and TX to Pin 7 (attempting to connect a Neo-6M-0-001 GPS module).

Observed Behavior:
- The app immediately shows "Error: Disconnected", without displaying the usual loading/progress bar, and then the device reboots.
- After reboot, the position settings are updated, but:
    - Sending a message results in "Queued for Sending" for a long time, followed by "No Interface".
    - Additional settings cannot be sent.
    - Attempting to reset RX/TX pins to 0 causes the settings progress bar to stay at 0%.
    - Waiting long enough (~5-10min) Reboots the Device, the GPS Pins are reset and message sending works again.

Hardware Context:
- The device is an ESP32-S3 variant with the Wio SX1262 module.
- Initially suspected a pin conflict due to shared B2B header usage or ESP32's IO multiplexing.
- However, tried multiple combinations (First D9/D10 then D6/D7), and the same issue occurs.
- According to [this forum post](https://forum.seeedstudio.com/t/using-the-wio-sx1262-for-xiao-to-easily-build-lora-point-to-point-communication-with-various-xiaos/291605/3), the pins should be available.

Any guidance on avoiding this issue or clarification how it's meant to work would be greatly appreciated.

Thanks a lot!

### Relevant log output

```Shell
DEBUG | 11:26:46 139 [Router] Save to disk 1

DEBUG | 11:26:46 139 [Router] Opening /prefs/config.proto, fullAtomic=1

INFO | 11:26:47 139 [Router] Save /prefs/config.proto

INFO | 11:26:47 139 [Router] Reboot in 7 seconds

DEBUG | 11:26:47 139 [Router] Module 'Admin' considered

DEBUG | 11:26:47 139 [Router] Module 'routing' wantsPacket=1

INFO | 11:26:47 139 [Router] Received routing from=0x0, id=0x956a644d, portnum=6, payloadlen=36

DEBUG | 11:26:47 139 [Router] Routing sniffing (id=0x956a644d fr=0x00000000 to=0xda73eb3c, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 rxtime=1746444406 priority=70)

DEBUG | 11:26:47 139 [Router] Partially randomized packet id 4192801509

DEBUG | 11:26:47 139 [Router] Enqueued local (id=0xf9e912e5 fr=0xda73eb3c to=0xda73eb3c, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=956a644d rxtime=1746444407 priority=120)

DEBUG | 11:26:47 139 [Router] Module 'routing' considered

DEBUG | 11:26:47 139 [Router] Rx someone rebroadcasting for us (id=0xf9e912e5 fr=0xda73eb3c to=0xda73eb3c, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=956a644d rxtime=1746444407 priorit

DEBUG | 11:26:47 139 [Router] Didn't find pending packet

DEBUG | 11:26:47 139 [Router] Add packet record fr=0xda73eb3c, id=0xf9e912e5

DEBUG | 11:26:47 139 [Router] handleReceived(REMOTE) (id=0xf9e912e5 fr=0xda73eb3c to=0xda73eb3c, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=956a644d rxtime=1746444407 priority=120)

DEBUG | 11:26:47 139 [Router] Module 'routing' wantsPacket=1

INFO | 11:26:47 139 [Router] Received routing from=0xda73eb3c, id=0xf9e912e5, portnum=5, payloadlen=2

DEBUG | 11:26:47 139 [Router] Routing sniffing (id=0xf9e912e5 fr=0xda73eb3c to=0xda73eb3c, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=956a644d rxtime=1746444407 priority=120)

DEBUG | 11:26:47 139 [Router] Received a ACK for 0x956a644d, stopping retransmissions

DEBUG | 11:26:47 139 [Router] Delivering rx packet (id=0xf9e912e5 fr=0xda73eb3c to=0xda73eb3c, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=956a644d rxtime=1746444407 priority=120)

DEBUG | 11:26:47 139 [Router] Update DB node 0xda73eb3c, rx_time=1746444407

DEBUG | 11:26:47 139 [Router] Forwarding to phone (id=0xf9e912e5 fr=0xda73eb3c to=0xda73eb3c, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=956a644d rxtime=1746444407 priority=120)

DEBUG | 11:26:47 139 [Router] Module 'routing' considered

DEBUG | 11:26:53 145 [Power] Battery: usbPower=1, isCharging=1, batMv=6368, batPct=100

INFO | 11:26:54 146 Rebooting
```

lagunacomputer, 2025-05-05:
if you can compile your own firmware, try adding  the following to your variant.h file.

#define GPS_UBLOX

or

#define GPS_UBLOX
#define GPS_RX_PIN 6
#define GPS_TX_PIN 7

Genei180, 2025-05-13:
This only helps somewhat.
The Pins are now Set RIght, but the Module still is not Detected.
Tryed with:
`#define GPS_DEBUG 1
#define HAS_GPS 1
#define GPS_UBLOX
#define GPS_RX_PIN 6
#define GPS_TX_PIN 7
#define GPS_BAUDRATE 9600 `
In the Variant.h

Log
`DEBUG | ??:??:?? 5 Pin EN HI
DEBUG | ??:??:?? 5 Use GPIO6 for GPS RX
DEBUG | ??:??:?? 5 Use GPIO7 for GPS TX
DEBUG | ??:??:?? 5 Start multicast thread
DEBUG | ??:??:?? 5 NeighborInfoModule is disabled
DEBUG | ??:??:?? 5 Rescan for I2C keyboard
DEBUG | ??:??:?? 5 Scan for I2C devices on port 1
DEBUG | ??:??:?? 5 Scan address 0x1f
DEBUG | ??:??:?? 6 Scan address 0x34
DEBUG | ??:??:?? 7 Scan address 0x55
DEBUG | ??:??:?? 8 Scan address 0x5a
DEBUG | ??:??:?? 9 Scan address 0x5f
INFO | ??:??:?? 10 External Notification Module Disabled
DEBUG | ??:??:?? 10 SX126xInterface(cs=41, irq=39, rst=42, busy=40)
DEBUG | ??:??:?? 10 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V
INFO | ??:??:?? 10 Start meshradio init
INFO | ??:??:?? 10 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO | ??:??:?? 10 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO | ??:??:?? 10 myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 MHz)
INFO | ??:??:?? 10 numChannels: 104 x 250.000kHz
INFO | ??:??:?? 10 channel_num: 20
INFO | ??:??:?? 10 frequency: 906.875000
INFO | ??:??:?? 10 Slot time: 28 msec
INFO | ??:??:?? 10 Set radio: final power level=22
INFO | ??:??:?? 21 SX126x init result -2
WARN | ??:??:?? 21 No SX1262 radio
INFO | ??:??:?? 21 Not using WIFI
ERROR | ??:??:?? 21 NOTE! Record critical error 3 at src/main.cpp:1281
INFO | ??:??:?? 21 PowerFSM init, USB power=1
DEBUG | ??:??:?? 21 State: BOOT
DEBUG | ??:??:?? 21 Free heap : 209612 bytes
DEBUG | ??:??:?? 21 Free PSRAM : 8347211 bytes
DEBUG | ??:??:?? 21 [Power] Battery: usbPower=0, isCharging=0, batMv=3100, batPct=0
DEBUG | ??:??:?? 21 [GPS] Probe for GPS at 9600
DEBUG | ??:??:?? 21 [GPS] Trying $PDTINFO (Unicore Family)...
DEBUG | ??:??:?? 21 [GPS]
DEBUG | ??:??:?? 21 [GPS] Trying $PCAS06,1*1A (ATGM33xx Family)...
DEBUG | ??:??:?? 22 [GPS]
DEBUG | ??:??:?? 22 [GPS] Trying $PAIR021*39 (Airoha Family)...
DEBUG | ??:??:?? 23 [GPS]
DEBUG | ??:??:?? 23 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 23 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 24 [GPS] Trying $PMTK605*31 (MTK Family)...
DEBUG | ??:??:?? 25 [GPS]
DEBUG | ??:??:?? 25 [GPS]
WARN | ??:??:?? 25 [GPS] No response for class 06 message 08
WARN | ??:??:?? 25 [GPS] No GNSS Module (baudrate 9600)
INFO | ??:??:?? 25 [RangeTest] Range Test Module - Disabled
INFO | ??:??:?? 25 [PowerFSM] Loss of power in Powered
INFO | ??:??:?? 25 [PowerFSM] Loss of power in Powered
INFO | ??:??:?? 25 [PowerFSM] Init the NimBLE bluetooth module
DEBUG | ??:??:?? 25 [PowerFSM] State: ON
DEBUG | ??:??:?? 27 [GPS] Probe for GPS at 9600
DEBUG | ??:??:?? 27 [GPS] Trying $PDTINFO (Unicore Family)...
DEBUG | ??:??:?? 28 [GPS]
DEBUG | ??:??:?? 28 [GPS] Trying $PCAS06,1*1A (ATGM33xx Family)...
DEBUG | ??:??:?? 29 [GPS]
DEBUG | ??:??:?? 29 [GPS] Trying $PAIR021*39 (Airoha Family)...
DEBUG | ??:??:?? 30 [GPS]
DEBUG | ??:??:?? 30 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 30 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 31 [GPS] Trying $PMTK605*31 (MTK Family)...
DEBUG | ??:??:?? 31 [GPS]
DEBUG | ??:??:?? 32 [GPS]
WARN | ??:??:?? 32 [GPS] No response for class 06 message 08
WARN | ??:??:?? 32 [GPS] No GNSS Module (baudrate 9600)
DEBUG | ??:??:?? 32 [GPS] Probe for GPS at 9600
DEBUG | ??:??:?? 32 [GPS] Trying $PDTINFO (Unicore Family)...
DEBUG | ??:??:?? 33 [GPS]
DEBUG | ??:??:?? 33 [GPS] Trying $PCAS06,1*1A (ATGM33xx Family)...
DEBUG | ??:??:?? 33 [GPS]
DEBUG | ??:??:?? 33 [GPS] Trying $PAIR021*39 (Airoha Family)...
DEBUG | ??:??:?? 34 [GPS]
DEBUG | ??:??:?? 34 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 35 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 35 [GPS] Trying $PMTK605*31 (MTK Family)...
DEBUG | ??:??:?? 36 [GPS]
DEBUG | ??:??:?? 37 [GPS]
WARN | ??:??:?? 37 [GPS] No response for class 06 message 08
WARN | ??:??:?? 37 [GPS] No GNSS Module (baudrate 9600)
WARN | ??:??:?? 37 [GPS] Give up on GPS probe and set to 9600`

Was Inside and the GPS had no Satellite Connection, maybe thats Part of the Problem?
But i atleast expect it to somehow communicate and Show up.

Side Note:
I tried with the Pins the other Way around, and when compiling over PlatformIO the GPS Pins Stayed in Flash Memory after Reupload.
So the Re-upload had no Effect on Changed Pins.
Had to Update it via, the Online Meshtastic Erase&Install, to be able to reflash it fully via VS Code.

Genei180, 2025-05-13:
I since found out the GPS Module was broken, because the GPS Neo 6M LED was never flashing even Outside.
Since then, I replaced it with a new Module, the LED Flashes now.

And the Arduino Sketch Provided here:
https://wiki.seeedstudio.com/get_start_l76k_gnss/
Works for the Neo 6M too.
I can log out GPS Data.

Unfortunately, Meshtastic doesn't recognize the GPS Module.
Any help or Guidance Moving Forward is appreciated.

lagunacomputer, 2025-05-15:
try 44/43 or 43/44 instead of 6/7 or 7/6

#define GPS_RX_PIN **44**
#define GPS_TX_PIN **43**

re:
https://www.seeedstudio.com/XIAO-ESP32S3-p-5627.html
and

[firmware](https://github.com/meshtastic/firmware/tree/master)/[variants](https://github.com/meshtastic/firmware/tree/master/variants)/[seeed_xiao_s3](https://github.com/meshtastic/firmware/tree/master/variants/seeed_xiao_s3)
/variant.h

/*Warning:
https://www.seeedstudio.com/L76K-GNSS-Module-for-Seeed-Studio-XIAO-p-5864.html
L76K Expansion Board can not directly used, L76K Reset Pin needs to override or physically remove it,
otherwise it will conflict with the SPI pins
*/
#define GPS_L76K
#ifdef GPS_L76K
**#define GPS_RX_PIN 44
#define GPS_TX_PIN 43**
#define HAS_GPS 1
#define GPS_BAUDRATE 9600
#define GPS_THREAD_INTERVAL 50
#define PIN_SERIAL1_RX PIN_GPS_TX
#define PIN_SERIAL1_TX PIN_GPS_RX
#define PIN_GPS_STANDBY 1
#endif

fifieldt, 2025-05-15:
Hi, just confirming that Ublox 6 is supported . 

To take some of the guesswork out, you can add
define GPS_DEBUG 1
in your variant.h
and that will print every message sent on the serial bus.

If you don't see any messages, there is probably a problem with your pins