meshtastic/firmware #6760: [Feature Request]: Use a stripped down version of nodedb for routers

RCGV1, 2025-05-07:
### Platform

Cross-Platform

### Description

Now with next hop nodedb size is important for router nodes to remember next hop information. Could we only store node id and next hop info in node db for router nodes so we can greatly expand the node db size.

RCGV1, 2025-05-07:
Or maybe a addition could be instead of hard limiting node db size arbitrarily we look at the actual space that node db is taking up and cap that space.

b8b8, 2025-05-07:
Would that then limit the "always retransmit" feature of a router to basically only group messages?

RCGV1, 2025-05-07:
Sorry not following, this only has to do with node db

b8b8, 2025-05-07:
Next hop data is stored in the sending node packet. If the routers participate in that routing then they won't forward all packets (DMs). They will only retransmit te packet of they are the assigned next hop.

RCGV1, 2025-05-07:
Thats not how next hop works. Everyone has to participate

b8b8, 2025-05-07:
Nvm, I see what you are saying 🙃