meshtastic/firmware #6766: [Bug]: Switching back from Router to Client keeps screen off

d4rkmen, 2025-05-08:
### Category

Other

### Hardware

T-Lora v2 1.6

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.7

### Description

Switching to Router role makes screen off as documented, but switching back to Client role - screen remains on.
This caused by rewriting config “Screen on for” value.
Setting back this value fixes the problem.

### Relevant log output

```Shell

```

dirksan28, 2025-05-09:
That could almost be a feature. (Just kidding.) 
But being able to turn off the display even in client mode is a feature/config option I sorely miss. There are also several Reddit posts asking for something like this... ;-)

d4rkmen, 2025-05-09:
For that particular case the workaround is to set screen on time = 10-15sec