meshtastic/firmware #6770: [Feature Request]: Rebroadcast mode: Combination out of "LOCAL_ONLY" and "CORE_PORTNUMS_ONLY" needed.

SR7673, 2025-05-08:
### Platform

NRF52

### Description

Hi programmers.
I'm running a ROUTER in a superior location. The mesh mostly has 80+ active nodes and i'm often getting 3...4 % AirUtilTX. Since the node must be as unobstrusive as possible, it has no solar module, just a 30 Ah accu...
Thus i want to limit the traffic to the most necessary stuff only. So far i choose LOCAL_ONLY since i only want to support the public channel and private messages between users but not private channels...
After upgrading to 2.6.5, i discovered the "CORE_PORTNUMS_ONLY" setting, which is a good idea and i would like to choose that but then i would allow traffic of other private channels.

So my request would be to build something that combines the limitations of  "LOCAL_ONLY" and "CORE_PORTNUMS_ONLY". That would be the most vavourable option for me.

Regards...

closed 2025-07-03 by github-actions[bot]

shalberd, 2025-08-26:
Agreed, a combination of those two would be great for 
a) avoiding spam by using CORE_PORTNUMS_ONLY, recently really well-done in Meshtastic Firmware 2.7.4.c1f4f79 Alpha by using a whitelist approach https://github.com/meshtastic/firmware/pull/7487/files
b) only the public channel / channel number 0 and direct messages between users as in LOCAL_ONLY

I can completely relate to the poster's wish as we have, on EU_868, a very limited frequency situation with one frequency slot 869.525 Mhz for all modem presets ... For mountain routers, this combination of the two roles would be awesome to have.

@thebentern what do you think?

ndoo, 2025-09-27:
+1, am admin of 2 hilltop routers serving a very busy city and I would like to deter commercial users with LOCAL_ONLY while mitigating the high ChUtil/Airtime with CORE_PORTNUMS_ONLY.

Maybe another implementation idea would be to use bitwise flags similar to position flags, to only allowlist specific portnums? Perhaps AND the flags with the core portnums to prevent accidentally killing basic functionality?

ianmcorvidae, 2026-02-18:
> Does anyone else think, based on the code https://github.com/meshtastic/firmware/blob/v2.7.15.567b8ea/src/modules/RoutingModule.cpp#L13
> 

You've misunderstood this code. That `maybePKI` is to identify packets that may be DMs, in order to exempt those messages from the filtering; at that point in the code, the `channel` value is not a channel index, but a channel hash, and PKI DMs use the value 0 specifically, while channels might use any value 0-255. A few lines below, you can see the LOCAL_ONLY behavior -- if a message is still encrypted (that is, it comes from _none of_ your locally configured channels), and it isn't a PKI DM, then drop it. So the description is basically accurate, and inaccurate in the opposite direction from what you were thinking.

> "Ignores observed messages from foreign meshes that are open or those which it cannot decrypt. Only rebroadcasts message on the nodes local primary / secondary channels."
> 

This is correct with regard to channel messages -- the one thing it's wrong about is that it does still rebroadcast a small subset of messages that it can't decrypt -- namely, that it will still rebroadcast messages if they appear to be DMs from and to nodes that are known to it.

Just to walk through the code you linked specifically, up through line 28:

* identify packets that may be PKI messages by checking if their channel hash is 0, they were not successfully decrypted as a channel message, and they have a `to` field that isn't just the broadcast address
* if the packet was not successfully decrypted (and therefore is either a PKI message or from a foreign mesh), check if the rebroadcast mode is LOCAL_ONLY or KNOWN_ONLY.
* if the rebroadcast mode matches and If it couldn't be a PKI message, ignore it and don't rebroadcast. if the rebroadcast mode matches and if it could be PKI, check that both the from and to nodes in the packet are present in our nodedb. If they aren't, ignore it and don't rebroadcast.
* finally, if we're in ham mode and the packet we just saw isn't, drop it and don't rebroadcast