meshtastic/firmware #6771: [Bug]: Unbounded Reception Hop Count Causes Node Database Churn in Large Mesh Networks

dirksan28, 2025-05-08:
### Category

Other

### Hardware

Heltec V3

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4.b89355f

### Description

### Description:
I live in an area where the mesh network extends over vast distances, resulting in my node "seeing" far more than 100 nodes (the current limit used in `NodeDB.cpp`. This behavior forces nodes to be cyclically removed and re-added. In my logs, I frequently observe entries being removed and then later reinserted (see [NodeDB.cpp#L1579](https://github.com/meshtastic/firmware/blob/216fbf23434a60d200e2519c002ed1d010febdea/src/mesh/NodeDB.cpp#L1579)). Consequently, the display remains on almost constantly, and the log is filled with messages about nodes being added and removed. 

This is particularly problematic when entries for “direct connections” are removed—even though I understand that the favorites functionality (assuming the client supports it) can help mitigate this. 
Additionally, I suspect that the frequent persistence of the node.db into SRAM could contribute to wear over time.

### Observations:  
The current implementation allows setting a maximum hop count for transmitting messages; however, there is no corresponding limit for message *reception*. Consequently, I repeatedly observe node.db entries with hop counts much higher than my local setting. I suspect that the number of nodes the device must manage rises exponentially with the hop count of received nodes.

For example, running a local test yields log entries such as:

```
INFO  | 18:37:37 1862 [Router] Node database full with 100 nodes and 23248 bytes free. Erasing oldest entry
INFO  | 18:42:39 2164 [Router] Node database full with 100 nodes and 22992 bytes free. Erasing oldest entry
INFO  | 18:43:40 2225 [Router] Node database full with 100 nodes and 22912 bytes free. Erasing oldest entry
INFO  | 18:49:28 2573 [Router] Node database full with 100 nodes and 22428 bytes free. Erasing oldest entry
INFO  | 18:52:07 2732 [Router] Node database full with 100 nodes and 22196 bytes free. Erasing oldest entry
INFO  | 18:54:15 2859 [Router] Node database full with 100 nodes and 22036 bytes free. Erasing oldest entry
INFO  | 18:57:52 3076 [Router] Node database full with 100 nodes and 21828 bytes free. Erasing oldest entry
```

### Proposed Fix:  
After reviewing the code, I wonder if adjusting the logic for adding nodes to the database might help in larger mesh networks. One idea is to refuse to add nodes that are farther away than the allowed maximum hop count. For example, inserting a check at [NodeDB.cpp#L1516](https://github.com/meshtastic/firmware/blob/216fbf23434a60d200e2519c002ed1d010febdea/src/mesh/NodeDB.cpp#L1516) such as:

```cpp
if ((mp.hop_start != 0 && mp.hop_limit <= mp.hop_start) &&
    ((mp.hop_start - mp.hop_limit) > config.lora.hop_limit)) {
    return;
}
```

This conditional would ensure that nodes whose hop difference exceeds the configured `config.lora.hop_limit` are not added to node.db. 
This would prevent nodes from being added to the `node.db` if they are further away than the configured max-hop-count.

**Alternative idea**  
Introduce a dedicated configuration option and protobuf field to independently control the maximum hop count for *receiving* nodes. This would allow finer control but would require broader changes across firmware and clients.

### Request for Developer Feedback:  
I would appreciate feedback from the development team on whether this suggestion is viable. Is this approach likely to have side effects, or could it potentially improve node management in large-scale meshes? If it seems promising, I’d be happy to test it locally.

### Relevant log output

```Shell

```

mverch67, 2025-05-09:
I don't think the proposed fix is a good idea. If people find out that their shown meshes are suddenly much smaller than in previous versions and that the reason is the hop limit, most of them will just set the hop limit to maximum which will of course make the overall situation of the mesh (e.g. congestion) much worse than before.

dirksan28, 2025-05-09:
> I don't think the proposed fix is a good idea. If people find out that their shown meshes are suddenly much smaller than in previous versions and that the reason is the hop limit, most of them will [...]

Hello Manuel thank you for reply, I understand your point. I had also seen this as a problem - though more from a product perspective. Because, in my view, it doesn’t really add much value to see a lot of nodes that I, as a user, can’t actually reach due to the max-hop count. It just looks good. ;-)

On the other hand, the node is busy performing tasks that aren’t necessary, using more memory than needed, and I fear that the hardware might wear out faster because the node.db is written to the SRAM every few minutes.