meshtastic/firmware #6785: [Bug]: Position Broadcast not respecting settings

joshbowyer, 2025-05-10:
### Category

Other

### Hardware

Seeed Card Tracker T1000-E

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.7

### Description

I have the position settings as follows:

GPS Update Interval: 60 seconds
Position Broadcast Interval: 300 seconds
Smart Position (when toggled on, I have tried off and on):
Minimum Distance: 10 meters
Minimum Time: 60 seconds

Regardless of any permutation of these settings, the node broadcasts its position every minute. After upgrading from 2.5.17 to 2.6.7, it is no longer actually every minute, but is sporadic and will be every minute for a few minutes, then it will wait 5 minutes, then 3, then every minute, etc. I am seeing this from another node, and logging with Meshsense. The T1000-E is sitting stationary on my desk and has 10 sats in view.

### Relevant log output

```Shell

```

garthvh, 2025-05-11:
Do you have a position log from the device showing that the positions being generated from the GPS are at the same location?  Sounds like the GPS has a bad fix and is bouncing around.

joshbowyer, 2025-05-11:
Hmm looking at the position log, youre right. But it always has at least 8 sats in view, and is near a window. Unless theres something that should be done to refine it, I guess Ill just have to increase the smart position minimums?

I will test again with smart position turned off, but it was doing the same thing earlier with it off as well. But I will test to make sure the setting actually stuck.

-------- Original Message --------

garthvh, 2025-05-11:
Yeah 10m is not real viable with a device GPS, the quality of the gps position is not really taken into account at all.

closed 2025-05-11 by garthvh