meshtastic/firmware #6795: [Bug]: Exactly one message gets through via UDP between an ESP32 and the Linux native client

EternityForest, 2025-05-11:
### Category

WiFi

### Hardware

Linux Native

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4

### Description

Put an ESP32 node with UDP Mesh enabled, then start the Flatpak version of the app on a Linux computer with no physical radio, so the two are just talking over UDP.

Send a message from one of them to the other.  It could be a channel message or a DM, it will reliably work exactly once.  All subsequent messages from one to the other will fail.

Close and restart the Linux side and one single message will again get through, there is no need to do anything on the ESP32 side to make it work, which is why I'm thinking the issue is with the Portduino version.

### Relevant log output

```Shell

```

thebentern, 2025-05-12:
I believe this is fixed in one of the subsequent alphas.

fifieldt, 2025-05-12:
Glad to hear it! To keep the bugs clean, let's open a new one for that new observation.

closed 2025-05-12 by fifieldt