meshtastic/firmware #6831: [Feature Request]: Make MQTT traceroute output consisitent with device output

robansaldo, 2025-05-17:
### Platform

Cross-Platform

### Description

When performing a traceroute if an intermediate node is not in the db of the device the MQTT output uses the name "Unknown" for that intermediate node, where the device display uses the node id in the form "Meshtastic abde". It would be helpful if the MQTT output also used this name.

garthvh, 2025-05-26:
Are you using json? The Array of nodes should not be different unless you are using the JSON output which has to be crafted manually as there are not really any good embedded json parsing libraries of an appropriate size
.

robansaldo, 2025-05-28:
Yes, I am using the JSON output. It looks like in src/serialization/MeshPacketSerializer.cpp that the longname is assumed to be "Unknown" and then if a node in the traceroute is in the nodedb that longname is substituted. I'm wondering if instead of "Unknown" that "Meshtastic abcd" could be used instead where "abcd" are the last 4 digits in the node id.