meshtastic/firmware #6835: [Bug]: SHT45  not data

shadyslim2018, 2025-05-17:
### Category

Hardware Compatibility, Other

### Hardware

Heltec V3

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4.b89355f

### Description

Environment
Device: Meshtastic Heltec v3 (ESP32-based)

Sensor: Adafruit Sensirion SHT45 (I²C)

Wiring:

SCL → Pin 42
SDA → Pin 41
VCC → 3.3V
GND → GND

Symptoms:

Sensor readings return nan% (humidity) and nan℃ (temperature)

No valid data received

Expected Behavior
The SHT45 should report valid temperature and humidity readings when queried over I²C.

Actual Behavior
The sensor fails to provide valid data, returning NaN (Not a Number) values instead.

Troubleshooting Steps Taken
Checked wiring connections (ensured no loose connections, correct pin assignments).

Verified power supply (3.3V is stable, GND is properly connected).

Tested with a different microcontroller ESP-32 C6 to confirm the sensor works. 

### Relevant log output

```Shell
ESP-ROM:esp32s3-20210327

Build:Mar 27 2021

rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)

SPIWP:0xee

mode:DIO, clock div:1

load:0x3fce3808,len:0x4bc

load:0x403c9700,len:0xbd8

load:0x403cc700,len:0x2a0c

entry 0x403c98d0

��@INFO | ??:??:?? 0

//\ E S H T /\ S T / C

INFO | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset

DEBUG | ??:??:?? 0 Filesystem files (909312/1572864 Bytes):

DEBUG | ??:??:?? 0 /prefs/channels.proto (57 Bytes)

DEBUG | ??:??:?? 0 /prefs/config.proto (201 Bytes)

DEBUG | ??:??:?? 0 /prefs/device.proto (114 Bytes)

DEBUG | ??:??:?? 0 /prefs/module.proto (100 Bytes)

DEBUG | ??:??:?? 0 /prefs/nodes.proto (557 Bytes)

DEBUG | ??:??:?? 0 /static/.gitkeep (0 Bytes)

DEBUG | ??:??:?? 0 /static/Logo_Black.svg.gz (602 Bytes)

DEBUG | ??:??:?? 0 /static/Logo_White.svg.gz (610 Bytes)

DEBUG | ??:??:?? 0 /static/apple-touch-icon.png.gz (3164 Bytes)

DEBUG | ??:??:?? 0 /static/favicon.ico.gz (2270 Bytes)

DEBUG | ??:??:?? 0 /static/icon.svg.gz (852 Bytes)

DEBUG | ??:??:?? 0 /static/index-b9vwykds.js.gz (222120 Bytes)

DEBUG | ??:??:?? 0 /static/index-wV0rbIjU.css.gz (16573 Bytes)

DEBUG | ??:??:?? 0 /static/index.html.gz (547 Bytes)

DEBUG | ??:??:?? 0 /static/mapbox-gl-DY02hNYD.js.gz (391221 Bytes)

DEBUG | ??:??:?? 0 /static/maplibre-gl-H6mDd47p.js.gz (210429 Bytes)

DEBUG | ??:??:?? 0 /static/robots.txt.gz (42 Bytes)

DEBUG | ??:??:?? 0 /static/site.webmanifest.gz (197 Bytes)

[ 779][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=41 scl=42 freq=100000

[ 793][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=17 scl=18 freq=100000

DEBUG | ??:??:?? 0 Power::lipoInit lipo sensor is not ready yet

DEBUG | ??:??:?? 0 Use analog input 1 for battery level

INFO | ??:??:?? 0 ADC config based on Two Point values and fitting curve coefficients stored in eFuse

INFO | ??:??:?? 0 Scan for i2c devices

[ 827][W][Wire.cpp:301] begin(): Bus already started in Master Mode.

DEBUG | ??:??:?? 0 Scan for I2C devices on port 2

[ 887][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1

INFO | ??:??:?? 0 SHT31 found at address 0x44

[ 901][W][Wire.cpp:301] begin(): Bus already started in Master Mode.

DEBUG | ??:??:?? 0 Scan for I2C devices on port 1

DEBUG | ??:??:?? 0 0x2 subtype probed in 2 tries

INFO | ??:??:?? 0 2 I2C devices found

DEBUG | ??:??:?? 0 acc_info = 0

INFO | ??:??:?? 0 S:B:43,2.6.4.b89355f

INFO | ??:??:?? 0 Build timestamp: 1743283504

DEBUG | ??:??:?? 0 Total heap: 294672

DEBUG | ??:??:?? 0 Free heap: 257196

DEBUG | ??:??:?? 0 Total PSRAM: 0

DEBUG | ??:??:?? 0 Free PSRAM: 0

DEBUG | ??:??:?? 0 NVS: UsedEntries 172, FreeEntries 458, AllEntries 630, NameSpaces 9

DEBUG | ??:??:?? 0 Setup Preferences in Flash Storage

DEBUG | ??:??:?? 0 Number of Device Reboots: 5

INFO | ??:??:?? 0 BLE OTA firmware version 0.3.0.87f0f2f

INFO | ??:??:?? 0 No WiFi OTA firmware available

INFO | ??:??:?? 0 Init NodeDB

[ 997][E][vfs_api.cpp:105] open(): /littlefs/static/static does not exist, no permits for creation

[ 1015][E][vfs_api.cpp:105] open(): /littlefs/prefs/db.proto does not exist, no permits for creation

INFO | ??:??:?? 1 Load /prefs/nodes.proto

INFO | ??:??:?? 1 Loaded /prefs/nodes.proto successfully

INFO | ??:??:?? 1 Loaded saved nodedatabase version 24, with nodes count: 2

INFO | ??:??:?? 1 Load /prefs/device.proto

INFO | ??:??:?? 1 Loaded /prefs/device.proto successfully

INFO | ??:??:?? 1 Loaded saved devicestate version 24

INFO | ??:??:?? 1 Load /prefs/config.proto

INFO | ??:??:?? 1 Loaded /prefs/config.proto successfully

INFO | ??:??:?? 1 Loaded saved config version 24

INFO | ??:??:?? 1 Load /prefs/module.proto

INFO | ??:??:?? 1 Loaded /prefs/module.proto successfully

INFO | ??:??:?? 1 Loaded saved moduleConfig version 24

INFO | ??:??:?? 1 Load /prefs/channels.proto

INFO | ??:??:?? 1 Loaded /prefs/channels.proto successfully

INFO | ??:??:?? 1 Loaded saved channelFile version 24

[ 1204][E][vfs_api.cpp:105] open(): /littlefs/prefs/uiconfig.proto does not exist, no permits for creation

ERROR | ??:??:?? 1 Could not open / read /prefs/uiconfig.proto

DEBUG | ??:??:?? 1 cleanupMeshDB purged 0 entries

DEBUG | ??:??:?? 1 Use nodenum 0xb03b92b4

DEBUG | ??:??:?? 1 Number of Device Reboots: 5

DEBUG | ??:??:?? 1 Expand short PSK #1

INFO | ??:??:?? 1 Wanted region 3, using EU_868

DEBUG | ??:??:?? 1 Coerce telemetry to min of 30 minutes on defaults

DEBUG | ??:??:?? 1 Save to disk 4

DEBUG | ??:??:?? 1 Opening /prefs/device.proto, fullAtomic=1

INFO | ??:??:?? 1 Save /prefs/device.proto

DEBUG | ??:??:?? 1 Use GPIO00 for button

DEBUG | ??:??:?? 1 SPI.begin(SCK=9, MISO=11, MOSI=10, NSS=8)

DEBUG | ??:??:?? 1 Use compiled/slipstreamed tzplaceholder

DEBUG | ??:??:?? 1 Set Timezone to GMT0

DEBUG | ??:??:?? 1 Read RTC time as 1

DEBUG | ??:??:?? 1 Start multicast thread

DEBUG | ??:??:?? 1 NeighborInfoModule is disabled

DEBUG | ??:??:?? 1 Rescan for I2C keyboard

DEBUG | ??:??:?? 1 Scan for I2C devices on port 2

DEBUG | ??:??:?? 1 Scan address 0x1f

DEBUG | ??:??:?? 1 Scan address 0x55

DEBUG | ??:??:?? 1 Scan address 0x5a

DEBUG | ??:??:?? 1 Scan address 0x5f

DEBUG | ??:??:?? 1 Scan for I2C devices on port 1

DEBUG | ??:??:?? 1 Scan address 0x1f

DEBUG | ??:??:?? 1 Scan address 0x55

DEBUG | ??:??:?? 1 Scan address 0x5a

DEBUG | ??:??:?? 1 Scan address 0x5f

INFO | ??:??:?? 1 External Notification Module Disabled

INFO | ??:??:?? 1 Turn on screen

DEBUG | ??:??:?? 1 haveGlyphs=1

DEBUG | ??:??:?? 1 waypoint wants a UI Frame

DEBUG | ??:??:?? 1 canned wants a UI Frame

DEBUG | ??:??:?? 1 SX126xInterface(cs=8, irq=14, rst=12, busy=13)

DEBUG | ??:??:?? 1 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V

INFO | ??:??:?? 1 Start meshradio init

INFO | ??:??:?? 1 Radio freq=869.525, config.lora.frequency_offset=0.000

INFO | ??:??:?? 1 Set radio: region=EU_868, name=LongFast, config=0, ch=0, power=27

INFO | ??:??:?? 1 myRegion->freqStart -> myRegion->freqEnd: 869.400024 -> 869.650024 (0.250000 MHz)

INFO | ??:??:?? 1 numChannels: 1 x 250.000kHz

INFO | ??:??:?? 1 channel_num: 1

INFO | ??:??:?? 1 frequency: 869.525024

INFO | ??:??:?? 1 Slot time: 28 msec

INFO | ??:??:?? 1 Set radio: final power level=22

INFO | ??:??:?? 1 SX126x init result 0

INFO | ??:??:?? 1 Frequency set to 869.525024

INFO | ??:??:?? 1 Bandwidth set to 250.000000

INFO | ??:??:?? 1 Power output set to 22
```

shadyslim2018, 2025-05-17:
Just test the esp32-c6 again and it's working fine.

[ 40100][V][esp32-hal-i2c-ng.c:262] i2cWrite(): i2c_master_transmit: bus=0 addr=0x44 handle=0x40821c64 size=1
[ 40119][V][esp32-hal-i2c-ng.c:308] i2cRead(): i2c_master_receive: bus=0 addr=0x44 handle=0x40821c64 size=6
📤 Publishing: {"temperature":26.15,"humidity":34.61,"rssi":-34,"ip":"192.168.1.150"}
✅ Publish SUCCESS

shadyslim2018, 2025-06-22:
I just replaced it with BME680, works fine now.

Stephan-2025, 2025-07-31:
Have the same Issue with a SHT40 as well as SHT41 - it only shows "NaN"

Stephan-2025, 2025-07-31:
> Having the same issue with the SHT40 Have you found a solution?
> 
> EDIT: seems like my issue is that the SHT40 is being recognized as a SHT31
> 
> EDIT2: Found the issue! 0xc8d is not recognized as an 4x in the i2c2wire scan port logic

Is there a solution to fix it by myself? Or only wait for a fix in the Meshtastic-Firmware?

deftdev, 2025-08-24:
Having the same issue with a SHT40 being incorrectly identified as a SHT31

> INFO  | ??:??:?? 2 DFRobot Rain Gauge found at address 0x1d
> INFO  | ??:??:?? 2 SHT31 found at address 0x44
> INFO  | ??:??:?? 2 2 I2C devices found

Firmware 2.7.5 on a rp2040-lora

notmarek, 2025-08-24:
It's because the serial numbers arent specified in the firmware and it falls back to sht31 by default

andreas-xa86, 2025-09-20:
https://github.com/meshtastic/firmware/issues/7522