meshtastic/firmware #6844: [Bug]:

egzumer, 2025-05-19:
### Category

Other

### Hardware

Heltec Mesh Node T114

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.7

### Description

Hi. This report won't be as structured as you may want, sorry for that. I have nothing more than educated guesses, but this seems to be a real issue, so I want to provide some feedback.

We have a mesh of ~16 nodes.  I have a node on a roof and a node in home. Most of the nodes in the mesh are closer to each other and I'm a bit further away. There is a roof node further away that I have a good connection with. We used to set my roof node and the other one to routers. This way we had a pretty reliable connection. Then came 2.6.

The next hop routing seems to not work very well. My home node only sees the roof node. When I try to send trace routes to another node that has good signal I loose around 50% or sometimes even 100% responses, same thing with direct messages, confirmations get lost. I tried client, router, router_late on the roof, doesn't matter. When I send messages directly from the roof node 100% always gets through. When I went to the roof with my home node and sent from there, the traceroutes were going out through the roof node and coming back directly, again close to 100% were coming back.
I switched back to 2.5 in router_late mode on the roof node (home node still on 2.6) and now I get 100% responses when I send from home.
This test was conducted on a good signal short path. But the same thing happens on the whole mesh around me. Since 2.6 came out the mesh performance for traceroutes and direct messages worsened.

I had similar issues when the roof node was set in client mode. It looks like next hop routing effectively sets its hops the same way as client routing. We have some roof nodes on the mesh, but most are nodes in buildings. I presume routing gets stuck between buildings. What 2.6 does it effectively prevents me from taking advantage of router/router_late setting.

### Relevant log output

```Shell

```

canadamadman, 2025-05-19:
This realy dosnt seem like bug  seems more like your not useing it corectly. Not supposed to be usieng reouter on your home. At all. Ever. Its for mountains. And everyone had moved to medium fast now.

egzumer, 2025-05-19:
It seems that with a mountain there would be exactly the same issue.

GUVWAF, 2025-05-19:
The only thing next-hop routing does is actually remembering how it previously got a route via "managed flooding" as used for broadcasts and DMs pre-2.6. So, if using a `ROUTER`/`ROUTER_LATE`, and it successfully delivered a packet via it, it would remember this for next time.

egzumer, 2025-05-19:
The issue is that my roof node drops packets that come from somebody else and it is not that easy to see what is going on on the mesh, but I looked at the code and I see some potential solution. What do you think about adding this to `NextHopRouter::perhapsRelay`

```cpp
(dest->has_hops_away && dest->hops_away == 0 && dest->snr > -17.5)
```

GUVWAF, 2025-05-22:
> The issue is that my roof node drops packets that come from somebody else

So you mean that your packets actually make it to the destination, but the ACK does not come back? Are really all nodes on 2.6?
You can also check which node is set as next-hop in received packets.

> What do you think about adding this to NextHopRouter::perhapsRelay

I'm not sure. `hops_away == 0` does not necessarily mean that you can reach it without any hops, it only means you can hear it directly. There may be many nodes that can hear another node directly, if they would all try to send at roughly the same time, the chance of collision increases.
We should try to find out if the issue is that the wrong next-hop is set, or something else is happening.

GUVWAF, 2025-05-22:
> Also I'm not sure where to look for next_hop ids

You'll need to run `meshtastic --noproto` to get the real debug logs of the device.

> It is somewhere in the middle between old and new routing.

Not exactly. In the "old" routing, clients will not rebroadcast again if they don't hear anyone doing so already. With next-hop routing and thus your proposal, they will always rebroadcast if the condition holds, so it may be even more harmful than the old routing.

egzumer, 2025-05-22:
> You'll need to run `meshtastic --noproto` to get the real debug logs of the device.

It trims output messages down to ~220 characters per line. I can't see the relevant info. Whatsup with that? It is heltec v3.

GUVWAF, 2025-05-24:
Ah, yeah. We might actually consider increasing that buffer.

egzumer, 2025-05-25:
I setup another node on my balcony. I had some issues, when I connected the node with a USB serial and opened COM port the node was becoming deaf to incoming packets (oddly enough, with usb connected but port closed, it worked fine). I managed to redirect all incoming packets to UDP port on my computer, but the routing has been working perfectly fine for 2 days now, I couldn't reproduce the issue. There were some periods when it worked fine before, so I'll keep on testing so more.

egzumer, 2025-05-26:
Retransmission is being cancelled by duplicated response message coming from another node.

https://github.com/meshtastic/firmware/blob/baefda213a0dcb28196ecad4c92170ca6eef773e/src/mesh/NextHopRouter.cpp#L86

```
<14>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Received traceroute from=0xda634b1c, id=0xb103ad5c, portnum=70, payloadlen=42
<14>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Route traced:
0x49edab8a --> 0x433debc4 (6.25dB) --> 0x33691cd0 (-15.75dB) --> 0xda634b1c (5.25dB)
...(-9.00dB) 0x433debc4 <-- (-17.25dB) 0x433b8310 <-- 0xda634b1c
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Module 'traceroute' considered
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Module 'routing' wantsPacket=1
<14>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Received routing from=0xda634b1c, id=0xb103ad5c, portnum=70, payloadlen=56
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Routing sniffing (id=0xb103ad5c fr=0xda634b1c to=0x49edab8a, WantAck=1, HopLim=2 Ch=0x0 Portnum=70 requestId=64a1e45d rxtime=1748286365 rxSNR=-9 rxRSSI=-117 hopStart=3)
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: cancelSending id=0x64a1e45d, removed=0
<14>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Relaying received message coming from 0
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Found existing packet record for fr=0xda634b1c,to=0x49edab8a,id=0xb103ad5c
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Add packet record fr=0xda634b1c, id=0xb103ad5c
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Setting next hop for packet with dest 49edab8a to 8a
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Setting next retransmission in 8000 msecs:
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]:  (id=0xb103ad5c fr=0xda634b1c to=0x49edab8a, WantAck=1, HopLim=1 Ch=0x0 Portnum=70 requestId=64a1e45d rxtime=1748286365 rxSNR=-9 rxRSSI=-117 hopStart=3 nextHop=0x8a relay=0xc4)
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Expand short PSK #1
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: Use AES128 key!
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: enqueue for send (id=0xb103ad5c fr=0xda634b1c to=0x49edab8a, WantAck=1, HopLim=1 Ch=0x8 encrypted len=83 rxtime=1748286365 rxSNR=-9 rxRSSI=-117 hopStart=3 nextHop=0x8a relay=0xc4 priority=80)
<15>1 - Meshtastic_ebc4 Router - - - ﻿[109]: txGood=8,txRelay=6,rxGood=28,rxBad=0
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: rx_snr found. hop_limit:1 rx_snr:-9.000000
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: rx_snr found in packet. Setting tx delay:784
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: Module 'routing' considered
<15>1 - Meshtastic_ebc4 RadioIf - - - ﻿[110]: Lora RX (id=0xb103ad5c fr=0xda634b1c to=0x49edab8a, WantAck=1, HopLim=2 Ch=0x8 encrypted len=69 rxSNR=-8.25 rxRSSI=-113 hopStart=3 relay=0x80)
<15>1 - Meshtastic_ebc4 RadioIf - - - ﻿[110]: Packet RX: 747ms
<15>1 - Meshtastic_ebc4 RadioIf - - - ﻿[110]: rx_snr found. hop_limit:1 rx_snr:-9.000000
<15>1 - Meshtastic_ebc4 RadioIf - - - ﻿[110]: rx_snr found in packet. Setting tx delay:448
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: Found existing packet record for fr=0xda634b1c,to=0x49edab8a,id=0xb103ad5c
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: Add packet record fr=0xda634b1c, id=0xb103ad5c
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: Ignore dupe incoming msg (id=0xb103ad5c fr=0xda634b1c to=0x49edab8a, WantAck=1, HopLim=2 Ch=0x8 encrypted len=69 rxSNR=-8.25 rxRSSI=-113 hopStart=3 relay=0x80)
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: cancelSending id=0xb103ad5c, removed=1
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: Incoming msg was filtered from 0xda634b1c
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: Use channel 0 (hash 0x8)
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: Expand short PSK #1
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: Use AES128 key!
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: decoded message (id=0xb103ad5c fr=0xda634b1c to=0x49edab8a, WantAck=1, HopLim=2 Ch=0x0 Portnum=70 requestId=64a1e45d rxSNR=-8.25 rxRSSI=-113 hopStart=3 relay=0x80)
<15>1 - Meshtastic_ebc4 Router - - - ﻿[110]: shouldFilterReceived (id=0xb103ad5c fr=0xda634b1c to=0x49edab8a, WantAck=1, HopLim=2 Ch=0x0 Portnum=70 requestId=64a1e45d rxSNR=-8.25 rxRSSI=-113 hopStart=3 relay=0x80)
<12>1 - Meshtastic_ebc4 Position - - - ﻿[111]: Ch. util >25%. Skip send
<15>1 - Meshtastic_ebc4 RadioIf - - - ﻿[111]: Lora RX (id=0xb103ad5c fr=0xda634b1c to=0x49edab8a, WantAck=1, HopLim=2 Ch=0x8 encrypted len=60 rxSNR=-7.5 rxRSSI=-114 hopStart=3 relay=0xd0)
<15>1 - Meshtastic_ebc4 RadioIf - - - ﻿[111]: Packet RX: 681ms
<15>1 - Meshtastic_ebc4 Router - - - ﻿[111]: Found existing packet record for fr=0xda634b1c,to=0x49edab8a,id=0xb103ad5c
<15>1 - Meshtastic_ebc4 Router - - - ﻿[111]: Add packet record fr=0xda634b1c, id=0xb103ad5c
<15>1 - Meshtastic_ebc4 Router - - - ﻿[111]: Ignore dupe incoming msg (id=0xb103ad5c fr=0xda634b1c to=0x49edab8a, WantAck=1, HopLim=2 Ch=0x8 encrypted len=60 rxSNR=-7.5 rxRSSI=-114 hopStart=3 relay=0xd0)
<15>1 - Meshtastic_ebc4 Router - - - ﻿[111]: Incoming msg was filtered from 0xda634b1c
<15>1 - Meshtastic_ebc4 Router - - - ﻿[111]: Use channel 0 (hash 0x8)
<15>1 - Meshtastic_ebc4 Router - - - ﻿[111]: Expand short PSK #1
<15>1 - Meshtastic_ebc4 Router - - - ﻿[111]: Use AES128 key!
<15>1 - Meshtastic_ebc4 Router - - - ﻿[111]: decoded message (id=0xb103ad5c fr=0xda634b1c to=0x49edab8a, WantAck=1, HopLim=2 Ch=0x0 Portnum=70 requestId=64a1e45d rxSNR=-7.5 rxRSSI=-114 hopStart=3 relay=0xd0)
<15>1 - Meshtastic_ebc4 Router - - - ﻿[111]: shouldFilterReceived (id=0xb103ad5c fr=0xda634b1c to=0x49edab8a, WantAck=1, HopLim=2 Ch=0x0 Portnum=70 requestId=64a1e45d rxSNR=-7.5 rxRSSI=-114 hopStart=3 relay=0xd0)
```

egzumer, 2025-05-27:
As the retrasmission I mean it doesn't pass the packet at all. It shedules the packet for sending, then next duplicate cancells the sending. It treats the packet the same way as client mode, but the node is in router_late mode.