meshtastic/firmware #6848: [Bug]: Unphone boot looping

jp-bennett, 2025-05-20:
### Category

Hardware Compatibility

### Hardware

unPhone

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.9

### Description

The unphone is regularly failing to boot, either hanging right after ``INFO  | ??:??:?? 2 Power to TFT Backlight`` or a few moments later during the touchscreen1 thread. This usually results in a rapid reboot loop. Possibly SPI contention?

### Relevant log output

```Shell

```

mverch67, 2025-05-20:
MUI on the other hand is working ok?

mverch67, 2025-05-20:
I cannot reproduce it. The oled style UI works as expected.

mverch67, 2025-05-20:
I pressed the reset button to test the startup, did this 20 times. Once or twice it rebooted at second 3:

```
DEBUG | ??:??:?? 3 Free heap  :  195344 bytes
DEBUG | ??:??:?? 3 Free PSRAM : 8308795 bytes
DEBUG | ??:??:?? 3 [Screen] Screen: Started...
DEBUG | ??:??:?? 3 [Screen] haveGlyphs=1
INFO  | ??:??:?? 3 [RangeTest] Range Test Module - Disabled

[15:30:21.842] Disconnected
[15:30:22.845] Connected to /dev/ttyACM0
��@------------------------------------------
stored messages (oldest first):
store[6] = May 20 2025 at 15:22:36
```

Now I've setup the environment to dump a stack trace of the crash, but it is not crashing anymore. So I guess it must be related to some print/log statements (e.g. in interrupt handler).

mverch67, 2025-05-20:
Got one:

```
assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))

Backtrace: 0x40377dc2:0x3fcb31f0 0x403817a1:0x3fcb3210 0x40388949:0x3fcb3230 0x4038279e:0x3fcb3360 0x4206e7fa:0x3fcb33a0 0x420729f1:0x3fcb33c0 0x42074cb7:0x3fcb34a0 0x42067012:0x3fcb34c0 0x42066f6e:0x3fcb3520 0x4206732e:0x3fcb3540 0x4206feb2:0x3fcb3570 0x42064fdd:0x3fcb3590 0x42019af5:0x3fcb35d0

x3fcb3230 0x4038279e:0x3fcb3360 0x4206e7fa:0x3fcb33a0 0x420729f1:0x3fcb33c0 0x42074cb7:0x3fcb34a0 0x42067012:0x3fcb34c0 0x42066f6e:0x3fcb3520 0x4206732e:0x3fcb3540 0x4206feb2:0x3fcb3570 0x42064fdd:0x3fcb3590 0x42019af5:0x3fcb35d0
0x40377dc2: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
0x403817a1: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
0x40388949: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
0x4038279e: xQueueSemaphoreTake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1549 (discriminator 1)
0x4206e7fa: lgfx::v1::spi::beginTransaction(int) at :?
0x420729f1: lgfx::v1::Touch_XPT2046::getTouchRaw(lgfx::v1::touch_point_t*, unsigned int) at ??:?
0x42074cb7: lgfx::v1::Panel_Device::getTouchRaw(lgfx::v1::touch_point_t*, unsigned int) at ??:?
0x42067012: TFTDisplay::getTouch(short*, short*) at :?
0x42066f6e: TouchScreenImpl1::getTouch(short&, short&) at :?
0x4206732e: TouchScreenBase::runOnce() at :?
0x4206feb2: concurrency::OSThread::run() at ??:?
0x42064fdd: loop() at ??:?
0x42019af5: loopTask(void*) at :?

```

mverch67, 2025-05-20:
Hmmm... this is what I found regarding this error:
[github espressif/arduino-esp32](https://github.com/espressif/arduino-esp32/discussions/7954#discussioncomment-5283183)

jp-bennett, 2025-06-22:
Turns out the bootloop I was seeing was from shooting past MAX_THREADS.