meshtastic/firmware #6851: [Feature Request]: The ink screen automatically selects the driver function based on the ID.

Quency-D, 2025-05-20:
### Platform

ESP32

### Description

Add a feature on the ink screen board that automatically identifies the ink screen model based on the ID and then automatically selects the relevant driver.
But I found that the code has been fixed based on macros and cannot be modified. Do you have any suggestions?
```c++
  //src\graphics\EInkDisplay2.cpp
        // Create GxEPD2 objects
    auto lowLevel = new EINK_DISPLAY_MODEL(PIN_EINK_CS, PIN_EINK_DC, PIN_EINK_RES, PIN_EINK_BUSY, *hspi);
    adafruitDisplay = new GxEPD2_BW<EINK_DISPLAY_MODEL, EINK_DISPLAY_MODEL::HEIGHT>(*lowLevel);

//src\graphics\EInkDisplay2.h
    GxEPD2_BW<EINK_DISPLAY_MODEL, EINK_DISPLAY_MODEL::HEIGHT> *adafruitDisplay = NULL;

todd-herbert, 2025-05-20:
How do you suggest identifying the display model?

Quency-D, 2025-05-20:
Is it possible based on the User ID?
[image]

todd-herbert, 2025-05-20:
Oh, I actually didn't realize until now that the SDA pin is bi-directional. In theory, it might be possible to detect based on the user id.

If possible, I would prefer to not do this though! Meshtastic uses this kind of detection for I2C and it seems like a headache! The puzzle becomes harder with each new display, and also more flash space is used to store drivers.

todd-herbert, 2025-05-20:
Is there a specific reason for this feature? Are you intending to switch the display model used on an existing board?

Quency-D, 2025-05-21:
Yes, one of the ink screens on the board has been discontinued, and I plan to replace it with a new one. If a firmware can be compatible with both the new and old screens, it will be more convenient to use.

Quency-D, 2025-05-21:
> Oh, I actually didn't realize until now that the SDA pin is bi-directional. In theory, it might be possible to detect based on the user id.
>
> If possible, I would prefer to not do this though! [...]

Okay, regarding this issue, we won't do recognition for all screens. There may be different models of screens for specific boards.

todd-herbert, 2025-05-21:
> Okay, regarding this issue, we won't do recognition for all screens. There may be different models of screens for specific boards.

This sounds like a good plan to me 👍

todd-herbert, 2025-05-21:
> Yes, one of the ink screens on the board has been discontinued, and I plan to replace it with a new one. If a firmware can be compatible with both the new and old screens, it will be more convenient to use.

Maybe we can do screen recognition for just the affected boards? It would be much easier to handle if we only need to detect between two options.

todd-herbert, 2025-05-21:
Are you able to announce which boards will be affected, and what the new screen will be? I understand that maybe it's not possible to give this information yet, but it would be helpful for planning how we can best detect the change.

Quency-D, 2025-05-21:
> Are you able to announce which boards will be affected, and what the new screen will be? I understand that maybe it's not possible to give this information yet, but it would be helpful for planning the easiest way to detect the change.

heltec_wireless_paper will switch from JD79656 to SSD1682. The driver IC of this new screen is SSD1682, and I am testing its display function.

Quency-D, 2025-05-21:
> Maybe we can do screen recognition for just the affected boards? It would be much easier to handle if we only need to detect between two options.

Yes, that's a great idea.

todd-herbert, 2025-05-21:
Possibly we can even detect between the two by reading the BUSY line after reset? If i remember correctly, I think one model is high and the other is low.

Quency-D, 2025-05-21:
I think reading ID is a more reliable method. JD79656 does not have this register and the value read is 0xff. The value read from SSD1682 immediately after hardware reset is 0x01.
[image]

Quency-D, 2025-05-21:
This is a sample code for reading, it's a bit messy and I haven't had time to organize it yet.
```c++
#if 1
      pinMode(PIN_EINK_RES, OUTPUT);
      digitalWrite(PIN_EINK_RES, LOW);
      delay(100);
      digitalWrite(PIN_EINK_RES, HIGH);
      delay(100);
{
    uint8_t out[1] = {66};
  pinMode(PIN_EINK_SCLK, OUTPUT); // SCK
pinMode(PIN_EINK_DC, OUTPUT); // SCK
  pinMode(PIN_EINK_CS, OUTPUT); // SCK

  if (PIN_EINK_DC >= 0) digitalWrite(PIN_EINK_DC, LOW);
  if (PIN_EINK_CS >= 0) digitalWrite(PIN_EINK_CS, LOW);
  delay(10);

//   test_write(0X2F);
    uint8_t cmd = 0x2F;
    pinMode(PIN_EINK_MOSI, OUTPUT);  
    digitalWrite(PIN_EINK_SCLK, LOW);
    for (int i = 0; i < 8; i++)
    {
      digitalWrite(PIN_EINK_MOSI, (cmd & 0x80) ? HIGH : LOW);
      cmd <<= 1;
      digitalWrite(PIN_EINK_SCLK, HIGH);
        delayMicroseconds(1);
      digitalWrite(PIN_EINK_SCLK, LOW);
        delayMicroseconds(1);
    }
  delay(10);

  if (PIN_EINK_DC >= 0) digitalWrite(PIN_EINK_DC, HIGH);
  
  pinMode(PIN_EINK_MOSI, INPUT_PULLUP);  

  for (uint16_t i = 0; i < 1; i++) {
    uint8_t d = 0;
    for (int8_t b = 7; b >= 0; b--) {
      digitalWrite(PIN_EINK_SCLK, LOW);   // 拉低 SCK
      delayMicroseconds(1);
      digitalWrite(PIN_EINK_SCLK, HIGH);  // 拉高 SCK
      delayMicroseconds(1);
      if (digitalRead(PIN_EINK_MOSI)) d |= (1 << b);  
    }
    out[i] = d;
  }
    LOG_INFO("User ID: %02X", out[0]);
  if (PIN_EINK_CS >= 0) digitalWrite(PIN_EINK_CS, HIGH);
}
#endif

Quency-D, 2025-05-22:
Although it is a status register, the "Chip ID" field will not be changed, which has been confirmed with the manufacturer.
Yes, the BUSY pin busy output levels of JD79656 and SSD1682 are different, but I feel that this solution is not as reliable as reading the "Chip ID".

[image]

[image]

todd-herbert, 2025-05-22:
Attempting to reading the chip ID is also slightly unreliable though, right? The response of JD79656 to the SPI command is not officially defined, I think?

todd-herbert, 2025-05-22:
But I suppose the details aren't so important. Main thing is that it sounds like we can handle this variation in display hardware with no issues.

todd-herbert, 2025-05-22:
Oh, sorry, I misunderstood the original purpose of the discussion!

I hadn't looked at that code for a while, it's definitely a bit messy. Maybe the least confusing would be to split `defined(HELTEC_WIRELESS_PAPER)` to a different macro. No strong preference from me though, whatever works!

```cpp
#elif defined(HELTEC_WIRELESS_PAPER) // and VM-E213 in future?

// Prevent model selection in platformio.ini
#ifdef EINK_DISPLAY_MODEL
#error This variant detects display model at boot
#endif

        // Start HSPI
        hspi = new SPIClass(HSPI);
        hspi->begin(PIN_EINK_SCLK, -1, PIN_EINK_MOSI, PIN_EINK_CS);

        // Pick GxEPD2 class
        if (/* detect display */)
                adafruitDisplay = new GxEPD2_BW<GxEPD2_213_FC1, GxEPD2_213_FC1::HEIGHT>(
                        GxEPD2_213_FC1(PIN_EINK_CS, PIN_EINK_DC, PIN_EINK_RES, PIN_EINK_BUSY, *hspi));
        else
                adafruitDisplay = new GxEPD2_BW<GxEPD2_213_BN, GxEPD2_213_BN::HEIGHT>(
                        GxEPD2_213_BN(PIN_EINK_CS, PIN_EINK_DC, PIN_EINK_RES, PIN_EINK_BUSY, *hspi));

        // Init GxEPD2
        adafruitDisplay->init();
        adafruitDisplay->setRotation(3);
```

todd-herbert, 2025-05-22:
Ahh sorry, I misunderstood the original reason for this discussion! 
I see the challenge you meant now: the GxEPD2 drivers do not have a shared base class. 

I'm not sure there's any particularly elegant solution. If I'm counting correctly it seems like the adafruitDisplay member is only accessed four times. It might be easiest to manually modify these four calls.

Quency-D, 2025-05-22:
> Ahh sorry, I misunderstood the original reason for this discussion! I see the challenge you meant now: the GxEPD2 drivers do not have a shared base class.
>
> I'm not sure there's any particularly [...]

Thanks a lot, this looks like a good idea, I'll take a look at it.

todd-herbert, 2025-05-23:
Oh yes you're right. Most of the ones in EInkDisplay2.cpp aren't relevant, because they're in that variant specific code inside `EInkDisplay::connect()`. I definitely missed the calls in EInkDynamicDisplay.cpp though.

The only other option (other than a total re-write) would be to make a wrapper class which manually passes through any methods used. This isn't a great option either, but I did get a proof of concept working today. Maybe it's a useful starting point.

If anyone can think a more elegant way to handle this, please comment!

```cpp
// Wrapper class for GxEPD2_BW

// Generic signature at build time, allowing display model to be detected at run-time
// Workaround for issue of GxEPD2_BW objects not having a shared base class
// Only exposes methods which we are actually using

template <typename DISPLAY_MODEL_1, typename DISPLAY_MODEL_2>
class GxEPD2_BW_MultiWrapper
{
public:
    void drawPixel(int16_t x, int16_t y, uint16_t color)
    {
        if (model == 1)
            model1->drawPixel(x, y, color);
        else
            model2->drawPixel(x, y, color);
    }

    bool nextPage()
    {
        if (model == 1)
            return model1->nextPage();
        else
            return model2->nextPage();
    }

    void hibernate()
    {
        if (model == 1)
            model1->hibernate();
        else
            model2->hibernate();
    }

    void init(uint32_t serial_diag_bitrate = 0)
    {
        if (model == 1)
            model1->init(serial_diag_bitrate);
        else
            model2->init(serial_diag_bitrate);
    }

    void init(uint32_t serial_diag_bitrate, bool initial, uint16_t reset_duration = 20, bool pulldown_rst_mode = false)
    {
        if (model == 1)
            model1->init(serial_diag_bitrate, initial, reset_duration, pulldown_rst_mode);
        else
            model2->init(serial_diag_bitrate, initial, reset_duration, pulldown_rst_mode);
    }

    void setRotation(uint8_t x)
    {
        if (model == 1)
            model1->setRotation(x);
        else
            model2->setRotation(x);
    }

    void setPartialWindow(uint16_t x, uint16_t y, uint16_t w, uint16_t h)
    {
        if (model == 1)
            model1->setPartialWindow(x, y, w, h);
        else
            model2->setPartialWindow(x, y, w, h);
    }

    void setFullWindow()
    {
        if (model == 1)
            model1->setFullWindow();
        else
            model2->setFullWindow();
    }

    int16_t width()
    {
        if (model == 1)
            return model1->width();
        else
            return model2->width();
    }

    int16_t height()
    {
        if (model == 1)
            return model1->height();
        else
            return model2->height();
    }

    void clearScreen(uint8_t value = 0xFF)
    {
        if (model == 1)
            model1->clearScreen();
        else
            model2->clearScreen();
    }

    void endAsyncFull()
    {
        if (model == 1)
            model1->endAsyncFull();
        else
            model2->endAsyncFull();
    }

    // Exposes methods of the GxEPD2_EPD object which is usually available as GxEPD2_BW::epd
    class Epd2Wrapper
    {
    public:
        bool isBusy() { return m_epd2->isBusy(); }
        GxEPD2_EPD *m_epd2;
    } epd2;

    // Constructor
    // Select driver by passing whichModel as 1 or 2
    GxEPD2_BW_MultiWrapper(uint8_t whichModel, int16_t cs, int16_t dc, int16_t rst, int16_t busy, SPIClass &spi)
    {
        assert(whichModel == 1 || whichmodel == 2);
        model = whichModel;
        LOG_DEBUG("GxEPD2_BW_MultiWrapper using driver %d", model);

        if (model == 1)
        {
            model1 = new GxEPD2_BW<DISPLAY_MODEL_1, DISPLAY_MODEL_1::HEIGHT>(DISPLAY_MODEL_1(cs, dc, rst, busy, spi));
            epd2.m_epd2 = &(model1->epd2);
        }
        else if (model == 2)
        {
            model = 2;
            model2 = new GxEPD2_BW<DISPLAY_MODEL_2, DISPLAY_MODEL_2::HEIGHT>(DISPLAY_MODEL_2(cs, dc, rst, busy, spi));
            epd2.m_epd2 = &(model2->epd2);
        }
    }

private:
    uint8_t model;
    GxEPD2_BW<DISPLAY_MODEL_1, DISPLAY_MODEL_1::HEIGHT> *model1;
    GxEPD2_BW<DISPLAY_MODEL_2, DISPLAY_MODEL_2::HEIGHT> *model2;
};

// Declaration
GxEPD2_BW_MultiWrapper<GxEPD2_213_FC1, GxEPD2_213_BN> *adafruitDisplay;

// Definition
adafruitDisplay = new GxEPD2_BW_MultiWrapper<GxEPD2_213_FC1, GxEPD2_213_BN>(/*1 or 2*/, PIN_EINK_CS, PIN_EINK_DC, PIN_EINK_RES, PIN_EINK_BUSY, *hspi);
```

Quency-D, 2025-05-24:
> > What do you think of designing an abstract base class for GxEPD2_BW?
> 
> It would definitely be ideal if there was an abstract base class for GxEPD2_BW, although I think that would mean a significant restructuring of the GxEPD2 library, right? It's not a task I would personally be enthusiastic about..

Yes, refactoring is a big task, so let's think about other options before deciding.