meshtastic/firmware #6852: [Bug]: meshtastic --set store_forward.is_server true Disables SNR and RSSI reporting on the python library

mikecarper, 2025-05-20:
### Category

Serial

### Hardware

Station G2

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.8

### Description

```
meshtastic --set store_forward.enabled true
meshtastic --set store_forward.is_server true
```

https://github.com/SpudGunMan/meshing-around/issues/147#issuecomment-2893223778

### Relevant log output

```Shell

```

mikecarper, 2025-05-20:
[image]
Screenshot of the difference between the settings

mikecarper, 2025-05-20:
This was done all over RF. It's a bug

mikecarper, 2025-05-20:
https://meshtastic.org/docs/configuration/module/store-and-forward-module/#is-server
Document it at a minimum?

mikecarper, 2025-05-20:
https://github.com/meshtastic/firmware/blob/master/src/modules/StoreForwardModule.cpp What would need to change to make this happen? Bots are getting more popular these days

GUVWAF, 2025-05-21:
It should be added to `packetHistory` in `historyAdd()` and set in the MeshPacket in `preparePayload()`.

mikecarper, 2025-05-21:
[meshtasticFirmware.6852.store_forward.SNR_RSSI.patch](https://github.com/user-attachments/files/20371678/meshtasticFirmware.6852.store_forward.SNR_RSSI.patch)
something like this?

GUVWAF, 2025-05-21:
Yes, looks good to me.

mikecarper, 2025-05-21:
Any chance this will get put into the firmware?

GUVWAF, 2025-05-22:
Yes, you can open a pull request and it will be considered. I think it's reasonable, there's quite some PSRAM available.

closed 2025-07-14 by github-actions[bot]