meshtastic/firmware #6854: [Board]: T-Deck pro (PCM Audio & A7682E 4G)

Szetya, 2025-05-20:
### SOC

ESP32

### Lora IC

SX1262

### Product Link

https://lilygo.cc/products/t-deck-pro

### Description

Sales have started although it is a bit slow.
I managed to get a PCM Audio version. I would appreciate support. First round of basic peripherals needed.
https://github.com/Xinyuan-LilyGO/T-Deck-Pro/tree/master

Thanks in advance!
I would appreciate your help with EPD. @caveman99  @todd-herbert

Szetya, 2025-05-21:
Okay.
- Display works with ZinggJM library even if I get warnings. :)
- LoRa radio works.
- GPS works.
- I2C scan finds devices (unknown I2C: 0x1A Touch CST328 and 0x28 Gyroscope BHI260AP) 
0x55 on the target is BQ27220 and it is incorrectly recognized as a T-Deck keyboard.
Keyboard is TCA8418, but meshtastic TCA8418Keyboard.cpp is unusable. :(
Battery voltage not measured by GPIO PIN. 

Extra lines in the setup() of the main.cpp file:
```cpp
#if defined(T_DECK_PRO)
    pinMode(BOARD_LORA_EN, OUTPUT);
    digitalWrite(BOARD_LORA_EN, HIGH);
    pinMode(BOARD_1V8_EN, OUTPUT);
    digitalWrite(BOARD_1V8_EN, HIGH);
    pinMode(BOARD_6609_EN, OUTPUT);
    digitalWrite(BOARD_6609_EN, HIGH);
    // T-Deck pro has all three SPI peripherals (E-ink, SD, LoRa) attached to the same SPI bus
    // We need to initialize all CS pins in advance otherwise there will be SPI communication issues
    // e.g. when detecting the SD card
    pinMode(LORA_CS, OUTPUT);
    digitalWrite(LORA_CS, HIGH);
    pinMode(SDCARD_CS, OUTPUT);
    digitalWrite(SDCARD_CS, HIGH);
    pinMode(PIN_EINK_CS, OUTPUT);
    digitalWrite(PIN_EINK_CS, HIGH);
    delay(100);
#endif
```
log:
```plaintext
//\ E S H T /\ S T / C

INFO  | ??:??:?? 2 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 2 Filesystem files (20480/3538944 Bytes):
DEBUG | ??:??:?? 2  /prefs/channels.proto (61 Bytes)
DEBUG | ??:??:?? 2  /prefs/config.proto (177 Bytes)
DEBUG | ??:??:?? 2  /prefs/device.proto (125 Bytes)
DEBUG | ??:??:?? 2  /prefs/module.proto (96 Bytes)
DEBUG | ??:??:?? 2  /prefs/nodes.proto (1406 Bytes)
DEBUG | ??:??:?? 2 Power::lipoInit lipo sensor is not ready yet
INFO  | ??:??:?? 2 Scan for i2c devices
DEBUG | ??:??:?? 2 Scan for I2C devices on port 1
INFO  | ??:??:?? 2 Device found at address 0x1a was not able to be enumerated
INFO  | ??:??:?? 2 Device found at address 0x28 was not able to be enumerated
INFO  | ??:??:?? 2 TCA8418 found at address 0x34
INFO  | ??:??:?? 2 T-Deck keyboard found at address 0x55
INFO  | ??:??:?? 2 QMI8658 found at address 0x6b
INFO  | ??:??:?? 2 3 I2C devices found
DEBUG | ??:??:?? 2 acc_info = 0
INFO  | ??:??:?? 2 S:B:54,2.6.9.c70fa0ef
DEBUG | ??:??:?? 2 Total heap: 279520
DEBUG | ??:??:?? 2 Free heap: 236620
DEBUG | ??:??:?? 2 Total PSRAM: 0
DEBUG | ??:??:?? 2 Free PSRAM: 0
DEBUG | ??:??:?? 2 NVS: UsedEntries 100, FreeEntries 530, AllEntries 630, NameSpaces 7
DEBUG | ??:??:?? 2 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 2 Number of Device Reboots: 110
INFO  | ??:??:?? 2 No BLE OTA firmware available
INFO  | ??:??:?? 2 No WiFi OTA firmware available
INFO  | ??:??:?? 2 Init NodeDB
INFO  | ??:??:?? 2 Load /prefs/nodes.proto
INFO  | ??:??:?? 3 Loaded /prefs/nodes.proto successfully
INFO  | ??:??:?? 3 Loaded saved nodedatabase version 24, with nodes count: 9
INFO  | ??:??:?? 3 Load /prefs/device.proto
INFO  | ??:??:?? 3 Loaded /prefs/device.proto successfully
INFO  | ??:??:?? 3 Loaded saved devicestate version 24
INFO  | ??:??:?? 3 Load /prefs/config.proto
INFO  | ??:??:?? 3 Loaded /prefs/config.proto successfully
INFO  | ??:??:?? 3 Loaded saved config version 24
INFO  | ??:??:?? 3 Load /prefs/module.proto
INFO  | ??:??:?? 3 Loaded /prefs/module.proto successfully
INFO  | ??:??:?? 3 Loaded saved moduleConfig version 24
INFO  | ??:??:?? 3 Load /prefs/channels.proto
INFO  | ??:??:?? 3 Loaded /prefs/channels.proto successfully
INFO  | ??:??:?? 3 Loaded saved channelFile version 24
ERROR | ??:??:?? 3 Could not open / read /prefs/uiconfig.proto
DEBUG | ??:??:?? 3 cleanupMeshDB purged 1 entries
DEBUG | ??:??:?? 3 Use nodenum 0x2f911680
DEBUG | ??:??:?? 3 Number of Device Reboots: 110
DEBUG | ??:??:?? 3 Expand short PSK #1
INFO  | ??:??:?? 3 Wanted region 2, using EU_433
DEBUG | ??:??:?? 3 Coerce telemetry to min of 30 minutes on defaults
DEBUG | ??:??:?? 3 Save to disk 4
DEBUG | ??:??:?? 3 Opening /prefs/device.proto, fullAtomic=1
INFO  | ??:??:?? 3 Save /prefs/device.proto
DEBUG | ??:??:?? 3 Use GPIO00 for button
DEBUG | ??:??:?? 3 SPI.begin(SCK=36, MISO=47, MOSI=33, NSS=3)
DEBUG | ??:??:?? 3 Use compiled/slipstreamed tzplaceholder
DEBUG | ??:??:?? 3 Set Timezone to GMT0
DEBUG | ??:??:?? 3 Read RTC time as 3
INFO  | ??:??:?? 3 GPS power state move from OFF to ACTIVE
DEBUG | ??:??:?? 3 Use GPIO44 for GPS RX
DEBUG | ??:??:?? 3 Use GPIO43 for GPS TX
DEBUG | ??:??:?? 3 Start multicast thread
DEBUG | ??:??:?? 3 NeighborInfoModule is disabled
ERROR | ??:??:?? 3 Could not open / read /prefs/cannedConf.proto
INFO  | ??:??:?? 3 CannedMessageModule is enabled
INFO  | ??:??:?? 3 External Notification Module Disabled
INFO  | ??:??:?? 3 Do EInk init
_PowerOn : 36001
_Update_Full : 1004394
DEBUG | ??:??:?? 4 initialized, 
DEBUG | ??:??:?? 4 determineMode(): 
DEBUG | ??:??:?? 4 refresh=FAST, reason=FLAGGED_DEMAND_FAST, frameFlags=0x9
DEBUG | ??:??:?? 4 Update E-Paper
_PowerOn : 36001
_Update_Part : 643001
DEBUG | ??:??:?? 5 done
WARN  | ??:??:?? 5 GxEPD2 version has not been modified to support async refresh; using fallback behavior. Please update lib_deps in variant's platformio.ini file
INFO  | ??:??:?? 5 Turn on screen
```

[image]

Szetya, 2025-05-21:
https://github.com/user-attachments/assets/5b513244-2cc3-44c3-9185-1b6fcd2de0c2

https://github.com/user-attachments/assets/cac0f0c4-9271-41df-9aea-0c45f7608d5e

Szetya, 2025-05-25:
BQ25896 works. I used XPowersLib since it was available in esp32s3_base.build_flags.

[image]

mr-tbot, 2025-07-07:
Hey all - compiled yesterday but noticed charging on the 4G variant isn't working - I have the I2S variant coming later this month - did anyone figure out the proper code modifications or if this is a 4G versus I2S variant issue?

Cheers!

Szetya, 2025-07-08:
> Hey all - compiled yesterday but noticed charging on the 4G variant isn't working - I have the I2S variant coming later this month - did anyone figure out the proper code modifications or if this is a 4G versus I2S variant issue?
> 
> Cheers!

Hi!

In principle there should be no problem or correlation between charging and the 4G/PCM(I2S) version. 
At the moment, the development is that neither 4G nor I2s voice is implemented. In the future it will probably be split in two as the hardware has common PINs.
See also the PR discussion for more information on the status. (The guys are very smart. I'm just a blur compared to them. 😂)

konomikitten, 2025-08-15:
I've flashed 2.7.5.ddd1499 seems to work okay, but after I send a message if I try to open the menu it won't come up. Tapping for unpausing the screen works but nothing else. Jack version.

mverch67, 2025-08-16:
> I've flashed 2.7.5.ddd1499 seems to work okay, but after I send a message if I try to open the menu it won't come up. Tapping for unpausing the screen works but nothing else. Jack version.

Try the user button

closed 2025-09-02 by mverch67