meshtastic/firmware #6859: GPS LR1110 built into ME25LS01 chip, in ME25LS01-4Y10TD variant it does not load location data

nextechDI, 2025-05-21:
Greetings, I am performing tests on ME25LS01-4Y10TD, so far the connection between devices through the LoRa network works well, sending messages, basic telemetry and the range test module too, in the next few days I will try with external sensors, I am using the meshtastic 2.6. 4 firmware, but I am writing because I am having problems getting the GPS module included in the ME25LS01 chip to work since they can not collect location data giving NaN results, the ME25LS01 uses the LR1110 chip that includes LoRa and GPS-GNSS, I'm using a passive GPS antenna number 1568R-A, in my test configuration I have two boards with ME25LS01 chip, configured in tracking mode, with Bluetooth disabled, in the EU_868 region, working on the first channel, with the range test module enabled on one device I also tested it with range test disabled, with GPS enable mode, the GPS update interval at 30 seconds, one of the devices has the default gps pin rx override and pin tx override and pin enable override settings and the other device has the latest settings use pin rx override = 33 - pin tx override = 34 - pin enable = 3.

But the GPS loads location data, I tried from the web client in the node tab, and from the CLI with the command meshtastic --port COMX --nodes , in both cases in the latitude longitude height section NaN appears, and checking the raw data in the web cient in the node tab all the data associated with position appears as 0.

Attachment 2 log files for the meshtastic command --port COM8 --dest !2856c997 --request-position --seriallog > C:\Users\hp\Desktop\gps_log.txt, the gps_log.txt file corresponds to one device and the gps_log1.txt file to the other device, please I ask for support to know if I am wrong in the configuration of the board with ME25LS01 chip, or if on the contrary it is a problem in the meshtastic firmware that cannot enable the GPS on the ME25LS01?

----------------------------------------------------------
Archivo gps_log.txt

[37m[32mINFO  [0m| ??:??:?? 26788 [SerialConsole] [32mClient wants config, nonce=1760180597[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mState: SERIAL[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mGot 5 files in manifest[0m
[37m[0m[32mINFO  [0m| ??:??:?? 26788 [SerialConsole] [32mStart API client config[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFromRadio=STATE_SEND_MY_INFO[0m
[37m[0m[32mINFO  [0m| ??:??:?? 26788 [SerialConsole] [32mgetFromRadio=STATE_SEND_UIDATA[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend My NodeInfo[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend device metadata[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend channels 8[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend config: device[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend config: position[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend config: power[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend config: network[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend config: display[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend config: lora[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend config: bluetooth[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend config: security[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend config: sessionkey[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: mqtt[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: serial[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: ext notification[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: store forward[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: range test[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: telemetry[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: canned message[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: audio[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: remote hardware[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: neighbor info[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: ambient lighting[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: detection sensor[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend module config: paxcounter[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend known nodes[0m
[37m[0m[32mINFO  [0m| ??:??:?? 26788 [SerialConsole] [32mnodeinfo: num=0x2856c997, lastseen=0, id=!2856c997, name=Meshtastic c997[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mSend known nodes[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mDone sending nodeinfo[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFromRadio=STATE_SEND_FILEMANIFEST[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFile: nodes.proto (373) bytes[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFromRadio=STATE_SEND_FILEMANIFEST[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFile: config.proto (181) bytes[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFromRadio=STATE_SEND_FILEMANIFEST[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFile: module.proto (119) bytes[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFromRadio=STATE_SEND_FILEMANIFEST[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFile: channels.proto (60) bytes[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFromRadio=STATE_SEND_FILEMANIFEST[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFile: device.proto (161) bytes[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFromRadio=STATE_SEND_FILEMANIFEST[0m
[37m[0m[32mINFO  [0m| ??:??:?? 26788 [SerialConsole] [32mConfig Send Complete[0m
Connected to radio
Sending position request to !2856c997 on channelIndex:0 (this could take a while)
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFromRadio=STATE_SEND_PACKETS[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mFromRadio=STATE_SEND_PACKETS[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mphone downloaded packet (id=0x226d53e3 fr=0xda802bd6 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 hopStart=3 relay=0xd6 priority=10)[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26788 [SerialConsole] [34mGot client heartbeat[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26789 [SerialConsole] [34mPACKET FROM PHONE (id=0x379b6f41 fr=0x00000000 to=0x2856c997, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP priority=70)[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26789 [SerialConsole] [34mlocalSend to channel 0[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26789 [SerialConsole] [34mAdd packet record fr=0x0, id=0x379b6f41[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26789 [SerialConsole] [34mSetting next hop for packet with dest 2856c997 to 97[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26789 [SerialConsole] [34mSetting next retransmission in 6842 msecs: [0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26789 [SerialConsole] [34m (id=0x379b6f41 fr=0x00000000 to=0x2856c997, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP nextHop=0x97 relay=0xd6 priority=70)[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26789 [SerialConsole] [34mExpand short PSK #1[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26789 [SerialConsole] [34mUse AES128 key![0m
[... 113 lines cut ...]
[36m[0m[34mDEBUG [0m| ??:??:?? 26842 [DeviceTelemetry] [34mhandleReceived(LOCAL) (id=0x890b87e4 fr=0xda802bd6 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=10)[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26842 [DeviceTelemetry] [34mNo modules interested in portnum=67, src=LOCAL[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26842 [DeviceTelemetry] [34mAdd packet record fr=0xda802bd6, id=0x890b87e4[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26842 [DeviceTelemetry] [34mExpand short PSK #1[0m
[... cut ...]
[36m[0m[34mDEBUG [0m| ??:??:?? 26843 [RadioIf] [34mCompleted sending (id=0x890b87e4 fr=0xda802bd6 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x43 encrypted len=50 hopStart=3 relay=0xd6 priority=10)[0m
Aborting due to: Timed out waiting for position
[37m[0m[32mINFO  [0m| ??:??:?? 26845 [SerialConsole] [32mDisconnect from phone[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26845 [SerialConsole] [34mPhoneAPI::close()[0m
[36m[0m[34mDEBUG [0m| ??:??:?? 26845 [SerialConsole] [34mState: ON[0m

-------------------------------------------------------------------

thebentern, 2025-05-21:
The Minew stamp does not have a _real_ GPS on board and Meshtastic does not currently support passive GNSS scanning modes on the LR series of LoRA radios and it is not on the immediate road map. That is something that we would be willing to review a pull request of, but have no immediate plans of implementing.

closed 2025-05-21 by thebentern

nextechDI, 2025-05-21:
The ME25LS01 has an optional circuit by design, which can use active antennas for the LR1110 of the ME25LS01, with an active antenna and its associated circuit in the ME25LS01 - LR1110 could this collect GPS data or would it still not work due to problems with the non-real GPS? What function does the supposed GPS of the LR1110 have?

nextechDI, 2025-05-21:
Thank you for your quick response and all your effort on the meshtastic project.

markbirss, 2025-05-21:
> The ME25LS01 has an optional circuit by design, which can use active antennas for the LR1110 of the ME25LS01, with an active antenna and its associated circuit in the ME25LS01 - LR1110 could this [...]

https://github.com/jgromes/RadioLib/blob/master/examples/LR11x0/LR11x0_GNSS_Autonomous_Position/LR11x0_GNSS_Autonomous_Position.ino