meshtastic/firmware #6910: [Feature Request]: Add CPU usage to host Metrics

marshalleq, 2025-05-28:
### Platform

Cross-Platform

### Description

I realise there can be a slight performance penalty for this which is why I think it should be more like an opt in toggle for debugging, however I'm interested to know how for example a high site node is performing and whether cpu is a bottleneck, in particular on the single core low power devices.

Would it be possible to get this added in somewhere?

Thanks.

mverch67, 2025-05-29:
I would expect the arduino loop() polling to eat up 100% CPU all the time and therefore doesn't let the FreeRTOS idle task run to count its idle time.

mikerjuk, 2025-06-04:
> I would expect the arduino loop() polling to eat up 100% CPU all the time and therefore doesn't let the FreeRTOS idle task run to count its idle time.

The loop() function is called by the idle task when using FreeRTOS under Arduino.

closed 2025-08-10 by github-actions[bot]