meshtastic/firmware #6914: [Board]: T-Lora Pager

Nasimovy, 2025-05-29:
### SOC

ESP32

### Lora IC

Semtech SX1262

### Product Link

https://lilygo.cc/products/t-lora-pager

### Description

- [ ] display ST7796
supported/implemented into meshtastic but with no MUI capabilities, lovyanGFX supports it

- [ ] IO expander XL9555 (0x20)
supported with https://github.com/lewisxhe/SensorLib

- [ ] codec ES8311 (0x18)

- [ ] IMU BHI260AP (0x28)
supported with https://github.com/lewisxhe/SensorLib

- [ ] RTC PCF85063A (0x51)
supported with https://github.com/lewisxhe/SensorLib

- [ ] powermanage BQ25896 (0x6B)

- [ ] gauge BQ27220 (0x55)

- [ ] keyboard TCA8418 (0x34)
supported/implemented into meshtastic but with a different key map and no shift capability

- [ ] haptic DRV2605 (0x5A)
supported/implemented into meshtastic with https://github.com/lewisxhe/SensorLib

mverch67, 2025-06-01:
Looks very similar to the components used in https://github.com/meshtastic/firmware/pull/6936 .

PlantDaddy, 2025-07-10:
I just got one in the mail and was planning on trying to add support for this as my second attempt at committing to yall. Is there a 101 guide on adding new device support?

Szetya, 2025-07-23:
Basic functions are working.
-Display
-LoRa
-Keyboard
-GPS
-Rotary encoder (settings in the configuration interface CLI, WEB, APP)
Charging, discharging also BQ27220 as in T-deck pro.
Battery 1300mAh see picture for online time and charge percentage (BT Off, ClientMute)

(T-beam + ST7796 shield version completed at the same time. The shield has been in the drawer for too long. :D )

[image]

[image]

[image]

Nasimovy, 2025-07-23:
> > [@Szetya](https://github.com/Szetya) [#6575](https://github.com/meshtastic/firmware/pull/6575) here is my pull request for the display shield
> 
> Yes, yes. However, with the new Base UI, it no longer fits on ESP32. :( I had to say goodbye to wifi and a few other things.

huh, it fits on my t-beam, which display driver are you using?

Szetya, 2025-07-23:
> huh, it fits on my t-beam, which display driver are you using?

Lovyan 1.2.0
For some reason the full one didn't work for me.

Szetya, 2025-07-23:
I don't miss WiFi. I have never used it.
In fact, I am looking more and more towards the nRF52840.
Reasonably low power consumption, nothing more. Minimal style.
However, "unfortunately" Lilygo is making increasingly beautiful devices with plastic injection moulded housings that I can't resist. (and the T-display S3 pro LR1121 is on its way) :D

Nasimovy, 2025-07-23:
> Lovyan 1.2.0 For some reason the full one didn't work for me.

im using a modified version of the ST7789 driver

Szetya, 2025-07-23:
> im using a modified version of the ST7789 driver

Soon I think it will be time to switch to smaller drivers. 
I have never been able to program at a professional level. I'm still happy when I manage to do a few things myself. ESP32 S3 1.47>>
[image]

jace, 2025-07-28:
I got one too, but I'm not sure where to start. @Szetya do you have a PR or fork I can checkout?

jstockdale, 2025-08-02:
@Szetya do you have the T-Lora Pager branch somewhere? I'd love to try it out, please. Thanks :)

Szetya, 2025-08-03:
> [@Szetya](https://github.com/Szetya) , im using the modified library that adapts the ST7798to work with a esp32, the s3 should work with lovyangfx, also support meshtastic mui?
> 
> Im still waiting on a tlora pager to do that.

I do not plan to use MUI. I have not tried it with ST7796. 
Yes, everything would be much easier with these new devices if meshtastic would support this display. 
TLora pager is a bit interesting because of the XL9555 gpio expander.

SlavTiger, 2025-08-12:
anyone have a guide to setting up meshtastic on the t lora pager? im a bit of a noob to these, meshcore worked out of box but there are no nodes in my area, but plenty of meshtastic so id like to jump ship. sadly im not much of a coder, i specialize more in wiring than in coding.

Szetya, 2025-08-13:
@mverch67 Manuel is already working on it. Just be patient! :)
https://github.com/meshtastic/firmware/pull/7613

lim3soft, 2025-09-02:
i have a Page too, and the only firmware i found in mestastic site is the nightly build. https://nightly.link/meshtastic/firmware/workflows/main_matrix/master

but i have no idea how to flash it. can you guide ?

> I succeeded to flash my T-pager with firmware as supplied after the PR being merged, and while playing around in VScode I also noticed a target "tlora-pager-tft" in the sources, is anyone working on this, it currently doesn't compile ('me' error or issues with the branch?
> 
> ```
> Compiling .pio/build/tlora-pager-tft/lib0e1/meshtastic-device-ui/source/graphics/LVGL/LVGLGraphics.cpp.o
> Compiling .pio/build/tlora-pager-tft/lib0e1/meshtastic-device-ui/source/graphics/OLED/OLEDViewController.cpp.o
> .pio/libdeps/tlora-pager-tft/meshtastic-device-ui/source/graphics/DeviceGUI.cpp:29:10: fatal error: ui.h: No such file or directory
> ```

lim3soft, 2025-09-03:
Cool, I managed to get my Pager flashed and I'm enjoying mesh now, thanks @willumpie82, superb. Althought it's not the perfect one, but usable, the mechanical scrolling is kinda off,  anyway, i will keep it as it is while waiting the official one.

mverch67, 2025-09-17:
@enimatek-nl 
The LR1121 has been add in [PR7956](https://github.com/meshtastic/firmware/pull/7956) and is / will be available in master or firmware 2.7.9