meshtastic/firmware #6956: [Feature Request]: Request: Support for LILYGO  T-Echo Lite in Meshtastic Firmware

robbo987, 2025-06-04:
### Platform

NRF52

### Description

Hey Meshtastic team,

I’d love to see both the T-Echo Lite fully supported in the Meshtastic firmware. These compact LoRa devices seem like a perfect fit for Meshtastic’s off-grid communication goals, and adding support would give the community more awesome hardware options!

Relevant Links

Product Page: [https://lilygo.cc/products/t-echo-lite](url)

GitHub: [https://github.com/[Xinyuan-LilyGO/T-Echo-Lite](https://github.com/Xinyuan-LilyGO/T-Echo-Lite)](url)

Thanks for checking this out! You all are doing amazing work, 

Cheers!

Szetya, 2025-06-04:
Something has started.

[image]

adrianTNT, 2025-06-06:
My 4 T-Echo Lite are bored and would also appreciate a Meshtastic firmware :)
Is there any code or work in progress to look at at least? :) 
I tried pasting the normal T-echo .UF2 file and it didn't like it.

Szetya, 2025-06-06:
> My 4 T-Echo Lite are bored and would also appreciate a Meshtastic firmware :) Is there any code or work in progress to look at at least? :) I tried pasting the normal T-echo .UF2 file and it didn't like it.

Try it. The GPS may not work. There is no HW_Vendor so it will show up as a normal t-echo.

[firmware.zip](https://github.com/user-attachments/files/20634098/firmware.zip)

adrianTNT, 2025-06-07:
Nice. **I installed it now. Thank you!**

Do you know if the **solar port and battery charging should work**? I am not sure if that is related to the firmwre or it is more hardware related and works permanently. 

What would it take for this firmware/device to be fully supported on Meshtastic website? I don't  understand if this is like a fork of official code or is something that needs to be checked and approved in the official Meshtastic repo?

Maybe **IF** the 2 T-Echo are very similar, it would be easier to keep firmware as one ("T-Echo / T-Echo Lite")?

Szetya, 2025-06-07:
I cannot guarantee that the GPS will work. Maybe yes, maybe no. :D

[firmware_gps.zip](https://github.com/user-attachments/files/20638595/firmware_gps.zip)

eabase, 2025-06-10:
@Szetya 
@adrianTNT 

Great!! Thanks for sharing this.

---

> I tested it now but I don't see any GPS data.

It will obviously only work if you have installed the GPS module `L76K`, or compatible.
https://github.com/Xinyuan-LilyGO/T-Echo-Lite/issues/2

Did you install this?

---
 
[image]

eabase, 2025-06-10:
@Szetya 

Do you mind sharing the source code for your firmware?
(Also as Adrian said, plase check, as the 2 FW links seem to be the same.)

Szetya, 2025-06-10:
> [@Szetya](https://github.com/Szetya)
> 
> Do you mind sharing the source code for your firmware? (Also as Adrian said, plase check, as the 2 FW links seem to be the same.)

Hello! Please be a little patient. My GPS module will arrive soon, so I don't want to practice blindly.
The two FWs are different, compare them to Total Commander.
[image]
For the code I had to modify an external directory so the situation is a bit more complicated, I should start a PR there first but I'm not a good coder and I have no experience with github.

adrianTNT, 2025-06-11:
> Did you install this? [GPS IMAGE]

Was the question for me? Yes, I tested with that GPS module installed.

But I am unsure if I should edit any settings under GPS pins in the Meshtastic app.
Do I type 1.13 and 1.15 in the 2 boxes of the Meshtastic app? It only accepts integer.

Compared to other devices: the normal T-Echo, that one has 0,0,0 in the 3 boxes. 
A Seeedstudio T1000-E has 0,0,43 in the 3 boxes

This table is from the T-Echo Lite docs
| GPS pin          | nRF52840 pin      |
| :------------------: | :------------------:|
| UART_RX      | IO 1.13                  |
| UART_TX      | IO 1.15                  |
| 1PPS      | IO 0.29                  |
| WAKE_UP      | IO 1.10                  |
| POWER_RT9080_EN      | IO 1.11                  |

[image]

Szetya, 2025-06-11:
Although still blind, I made 2 attempts.

[GPS_test_firmware.zip](https://github.com/user-attachments/files/20692198/GPS_test_firmware.zip)

adrianTNT, 2025-06-17:
Thank you. I will test it soon. 

I also tested the previous ones that contained 2 files (`firmware_gps_RxRxSwapped.uf2` and `firmware_gps_RxRxUnswapped`) but it didn't work, I didn't mention it because I was thinking I am doing something wrong at my end and waited for others testers 😬

@Szetya does this node have any special **solar functions**? e.g can it show in the chart when solar is charging battery and when not? Or show voltage chart more accurately while charging (solar or not)? I seen most devices just show `100%` while charging, would be nice to see progress.

Szetya, 2025-06-17:
> > [@Szetya](https://github.com/Szetya)
> > > [T-Echo-lite_GPS_OK.zip](https://github.com/user-attachments/files/20759925/T-Echo-lite_GPS_OK.zip)
> > 
> > 
> > Can you please provide the full source code for this? I would like to add/change a few minor things. (Only looking at the UF2 binary HEX values are not very helpful.)
> 
> Will you do the PR? Need two, need to add the display to the meshtastic/GxEPD2 library . I don't know how to use github and I don't intend to understand it. 😉

[t-echo-lite.zip](https://github.com/user-attachments/files/20778362/t-echo-lite.zip)

eabase, 2025-06-19:
@Szetya 
> The code for meshtastic is here 

Yes, of course, I understand that it sounds like a stupid question, but I though the firmware needed to be modified separately for each device? 

The structure here is very confusing when you're not already used to developing in this framework. 

For example:
https://github.com/meshtastic/firmware/tree/master/variants/t-echo
https://github.com/meshtastic/firmware/blob/master/boards/t-echo.json

And the following doesn't really help explaining how the parts/libraries are connected. 
https://meshtastic.org/docs/development/firmware/build/

Szetya, 2025-06-22:
2.7.1 version available.

[firmware_2_7_1.zip](https://github.com/user-attachments/files/20853239/firmware_2_7_1.zip)

eabase, 2025-06-23:
@Szetya 
> 2.7.1 version available.

Do you have a changelog?

Also add here:
https://github.com/Xinyuan-LilyGO/T-Echo-Lite

Szetya, 2025-06-23:
> [@Szetya](https://github.com/Szetya)
> 
> > 2.7.1 version available.
> 
> Do you have a changelog?
https://github.com/meshtastic/firmware/releases

jstockdale, 2025-07-01:
@Szetya can you share your working branch or what modifications were necessary? I’m happy to help out with the PR or clean up anything necessary.

I appreciate all your hard work and thanks to you two of my devices are up and running instead of collecting dust.

Please don’t let any distractors take away from your contribution here. Thanks again!

jstockdale, 2025-07-01:
@eabase shame on you. Do you really think this firmware is *trojan’d*?!? To attack all of the two dozen people that probably have this device?

That’s absurd. If you want to decompile it in Ghidra I’d invite you to go ahead and find any anomalies. Otherwise, frankly, your attitude is damaging and if I were on the receiving end of your comments I probably would also get spicy in my replies not to mention *stop contributing to the project* …

Most folks out there aren’t getting anything from contributing time, code, and effort to this and many other open source projects. A good rule is: Don’t be an ass to folks trying to help out.

Szetya, 2025-07-01:
> [@Szetya](https://github.com/Szetya)Meg tudnád osztani a működő ágadat, vagy milyen módosításokra volt szükség? Örömmel segítek a PR-ben, vagy kijavítok bármit, ami szükséges.
> 
> Köszönöm a kemény munkátokat, és neked köszönhetően két készülékem is működik, ahelyett, hogy porosodna.
> 
> Kérlek, ne hagyd, hogy bármi is elvonja a figyelmedet a hozzászólásodról. Még egyszer köszönöm!

Thank you for your trust.
Time and energy spent and NOT knowing github is the reason why there is no PR.
There are pro people here who shake it out of their fingers. I thought eabase was like that but all I got was worming :D
Yes, I've attached all the necessary files above. If you can translate it, it will be a snap. ;)

Szetya, 2025-07-11:
Well, the first step should be to do that. 
https://github.com/meshtastic/GxEPD2/pull/13
But right now I have no idea what's going on. That's why I didn't want all this.

adrianTNT, 2025-08-06:
Hi @Szetya 

- what is latest firmware I could use?

- Also, with previous version I downloaded some weeks ago, I think the node goes in some kind of power saving? Have you (or anyone) experienced this? When the node is not accessed for some hours, when I do a traceroute from another node, it needs 1-2 minutes to reply the first time, then it responds faster on next tries.

Szetya, 2025-08-07:
> Hi [@Szetya](https://github.com/Szetya)
> 
> * what is latest firmware I could use?
> * Also, with previous version I downloaded some weeks ago, I think the node goes in some kind of power saving? Have you (or anyone) experienced this? When the node is not accessed for some hours, when I do a traceroute from another node, it needs 1-2 minutes to reply the first time, then it responds faster on next tries.
> 
> // I suspect that is some power saving / sleep also because this solar node is always at +95% battery with just a **tiny solar panel** 110x60 mm 😬

It's been maybe a month since I last updated. 
But I don't use it regularly so I can't speak from experience. 
I didn't change the functional part of meshtastic, the base is t-echo so it's likely that the power saving part might be related to that. 
My changes only concern the PIN configuration and the display definition.

bot123456-hud, 2025-08-14:
hi@Szetya 
Thank you for your contribution to creating the T-Echo Lite firmware.

I just started learning programming completely from scratch because of Meshtastic, so I don’t understand anything yet.

I’d like to ask if you have time to write a tutorial on where the files inside the T-Echo-Lite ZIP archive should be placed. If someone else could help reply on behalf of Szetya, I’d also be happy, because I don’t want to waste Szetya’s time with a simple question. I have confirmed that all the files required by the programming environment are present.

I placed it in firmware/variants/nrf52840/t-echo-lite, but it keeps showing this error:

27 | #include "variant.h"
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\t-echo-lite\lib81b\Adafruit_TinyUSB_Arduino\arduino\Adafruit_USBH_Host.cpp.o] Error 1
In file included from C:\Users\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5/Arduino.h:60, 
                 from C:\Users\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Adafruit_TinyUSB_Arduino\src\arduino\ports\nrf\Adafruit_TinyUSB_nrf.cpp:32: 
C:\Users\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5/Uart.h:27:10: fatal error: variant.h: No such file or directory

caveman99, 2025-08-15:
PR up at https://github.com/meshtastic/firmware/pull/7636

fifieldt, 2025-09-02:
PR was merged.

CryoRig, 2025-09-10:
Did someone get GPS to work?
the pinout from lilygo is very confusing (at least for me as there there are two different ones) and the one they use in their example firmware does not match with the one in this PR...
Pin definition from PR:
| Function | PIN |
|-|-|
|PIN_GPS_STANDBY| (32 + 10)|
|PIN_GPS_PPS| 0 + 29|
|GPS_TX_PIN| 32 + 15|
|GPS_RX_PIN| 32 + 13|

Pin definition from Lilygo: [source](https://github.com/Xinyuan-LilyGO/T-Echo-Lite/blob/9bad446749727a74f5b8387dc425a4886f813fda/libraries/Private_Library/t_echo_lite_config.h#L99)
| Function | PIN |PIN translated|
|-|-|-|
|GPS_WAKE_UP| EXT_1X7P_IO_1_13 | 32+13|
|GPS_1PPS| EXT_1X7P_IO_1_15 | 32+15|
|GPS_UART_TX| EXT_1X7P_IO_0_29 | 0+29|
|GPS_UART_RX| EXT_1X7P_IO_1_10 | 32+10|
|GPS_RT9080_EN| EXT_1X7P_IO_1_11 | 32+11|

hylaride, 2025-09-27:
Since it's still not showing up in the web flasher as of Sept 27, 2025, you can get the firmware by going to the [releases](https://github.com/meshtastic/firmware/releases) page, expand the assets for the version you want, and download firmware-nrf52840-XXXX.zip, where XXXX is the version number.  Within that zip will have a uf2 file called firmware-t-echo-lite-XXXX.uf2 that can be used to flash.