meshtastic/firmware #6965: [Bug]: esp32 WebServer stops serving API endpoints during config stream

philon-, 2025-06-05:
### Category

WiFi

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

v2.6.10.9ce4455

### Description

The ESP32 web server implementation stops serving API endpoints in the middle of a config stream, which results in incomplete config saves and unexpected web client behaviour.

This is in relation to an [issue](https://github.com/meshtastic/web/issues/633) in the web UI repo where an user attempted to disable WiFi while connected over HTTP.

Currently, the web server [`handleWebResponse()`](https://github.com/meshtastic/firmware/blob/c0e1616382a5493881a798e26114ba0f687a1e2a/src/mesh/http/WebServer.cpp#L59) checks [`isWifiAvailable()`](https://github.com/meshtastic/firmware/blob/c0e1616382a5493881a798e26114ba0f687a1e2a/src/mesh/wifi/WiFiAPClient.cpp#L209) which reads from memory (`config.network.wifi_enabled` or `config.network.eth_enabled`) which means that the admin message `commitEditSettings` cannot be sent to the device, committing the settings to flash and rebooting as expected.

### Relevant log output

```Shell
[HTTPS:I] Request: PUT /api/v1/toradio (FID=52)
DEBUG | 18:01:15 315 [WebServer] webAPI handleAPIv1ToRadio
DEBUG | 18:01:15 315 [WebServer] Received 30 bytes from PUT request
DEBUG | 18:01:15 315 [WebServer] PACKET FROM PHONE (id=0x0d0626a2 fr=0x67893018 to=0x67893018, WantAck=1, HopLim=0 Ch=0x0 Portnum=6 WANTRESP)
DEBUG | 18:01:15 315 [WebServer] Enqueued local (id=0x0d0626a2 fr=0x00000000 to=0x67893018, WantAck=1, HopLim=0 Ch=0x0 Portnum=6 WANTRESP rxtime=1749146475)
DEBUG | 18:01:15 315 [WebServer] webAPI handleAPIv1ToRadio
INFO  | 18:01:15 315 Tell client we have new packets 9
DEBUG | 18:01:15 315 [Router] Add packet record fr=0x0, id=0xd0626a2
DEBUG | 18:01:15 315 [Router] handleReceived(REMOTE) (id=0x0d0626a2 fr=0x00000000 to=0x67893018, WantAck=1, HopLim=0 Ch=0x0 Portnum=6 WANTRESP rxtime=1749146475)
DEBUG | 18:01:15 315 [Router] Module 'Admin' wantsPacket=1
INFO  | 18:01:15 315 [Router] Received Admin from=0x0, id=0xd0626a2, portnum=6, payloadlen=3
INFO  | 18:01:15 315 [Router] Handle admin payload 64
INFO  | 18:01:15 315 [Router] Begin transaction for editing settings
DEBUG | 18:01:15 315 [Router] Partially randomized packet id 4142543446
INFO  | 18:01:15 315 [Router] Asked module 'Admin' to send a response
DEBUG | 18:01:15 315 [Router] Module 'routing' wantsPacket=1
INFO  | 18:01:15 315 [Router] Received routing from=0x0, id=0xd0626a2, portnum=6, payloadlen=3
DEBUG | 18:01:15 315 [Router] Routing sniffing (id=0x0d0626a2 fr=0x00000000 to=0x67893018, WantAck=1, HopLim=0 Ch=0x0 Portnum=6 WANTRESP rxtime=1749146475)
DEBUG | 18:01:15 315 [Router] Another module replied to this message, no need for 2nd ack
DEBUG | 18:01:15 315 [Router] Module 'routing' considered
DEBUG | 18:01:15 315 [Router] Send response (id=0xf6ea3256 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d0626a2 rxtime=1749146475 priority=120)
DEBUG | 18:01:15 315 [Router] Update DB node 0x67893018, rx_time=1749146475
DEBUG | 18:01:15 315 [Router] Enqueued local (id=0xf6ea3256 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d0626a2 rxtime=1749146475 priority=120)
DEBUG | 18:01:15 315 [Router] Rx someone rebroadcasting for us (id=0xf6ea3256 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d0626a2 rxtime=1749146475 priority
DEBUG | 18:01:15 315 [Router] Didn't find pending packet
DEBUG | 18:01:15 315 [Router] Add packet record fr=0x67893018, id=0xf6ea3256
DEBUG | 18:01:15 315 [Router] handleReceived(REMOTE) (id=0xf6ea3256 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d0626a2 rxtime=1749146475 priority=120)
DEBUG | 18:01:15 315 [Router] Module 'routing' wantsPacket=1
INFO  | 18:01:15 315 [Router] Received routing from=0x67893018, id=0xf6ea3256, portnum=5, payloadlen=2
DEBUG | 18:01:15 315 [Router] Routing sniffing (id=0xf6ea3256 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d0626a2 rxtime=1749146475 priority=120)
DEBUG | 18:01:15 315 [Router] Received a ACK for 0xd0626a2, stopping retransmissions
DEBUG | 18:01:15 315 [Router] Delivering rx packet (id=0xf6ea3256 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d0626a2 rxtime=1749146475 priority=120)
DEBUG | 18:01:15 315 [Router] Update DB node 0x67893018, rx_time=1749146475
DEBUG | 18:01:15 315 [Router] Forwarding to phone (id=0xf6ea3256 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d0626a2 rxtime=1749146475 priority=120)
DEBUG | 18:01:15 315 [Router] Module 'routing' considered
INFO  | 18:01:15 315 Tell client we have new packets 11
[HTTPS:I] Request: GET /api/v1/fromradio?all=false (FID=52)
DEBUG | 18:01:16 315 [WebServer] webAPI handleAPIv1FromRadio
DEBUG | 18:01:16 315 [WebServer] FromRadio=STATE_SEND_PACKETS
DEBUG | 18:01:16 315 [WebServer] webAPI handleAPIv1FromRadio, len 11
[HTTPS:I] Request: GET /api/v1/fromradio?all=false (FID=52)
[... 25 lines cut ...]
INFO  | 18:01:16 316 [Router] Delay save of changes to disk until the open transaction is committed
DEBUG | 18:01:16 316 [Router] Partially randomized packet id 3117454935
INFO  | 18:01:16 316 [Router] Asked module 'Admin' to send a response
DEBUG | 18:01:16 316 [Router] Module 'routing' wantsPacket=1
INFO  | 18:01:16 316 [Router] Received routing from=0x0, id=0x2acf29d6, portnum=6, payloadlen=51
DEBUG | 18:01:16 316 [Router] Routing sniffing (id=0x2acf29d6 fr=0x00000000 to=0x67893018, WantAck=1, HopLim=0 Ch=0x0 Portnum=6 WANTRESP rxtime=1749146476)
DEBUG | 18:01:16 316 [Router] Another module replied to this message, no need for 2nd ack
DEBUG | 18:01:16 316 [Router] Module 'routing' considered
DEBUG | 18:01:16 316 [Router] Send response (id=0xb9d09657 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=2acf29d6 rxtime=1749146476 priority=120)
DEBUG | 18:01:16 316 [Router] Update DB node 0x67893018, rx_time=1749146476
DEBUG | 18:01:16 316 [Router] Enqueued local (id=0xb9d09657 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=2acf29d6 rxtime=1749146476 priority=120)
DEBUG | 18:01:16 316 [Router] Rx someone rebroadcasting for us (id=0xb9d09657 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=2acf29d6 rxtime=1749146476 priorit
DEBUG | 18:01:16 316 [Router] Didn't find pending packet
DEBUG | 18:01:16 316 [Router] Add packet record fr=0x67893018, id=0xb9d09657
DEBUG | 18:01:16 316 [Router] handleReceived(REMOTE) (id=0xb9d09657 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=2acf29d6 rxtime=1749146476 priority=120)
DEBUG | 18:01:16 316 [Router] Module 'routing' wantsPacket=1
INFO  | 18:01:16 316 [Router] Received routing from=0x67893018, id=0xb9d09657, portnum=5, payloadlen=2
DEBUG | 18:01:16 316 [Router] Routing sniffing (id=0xb9d09657 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=2acf29d6 rxtime=1749146476 priority=120)
DEBUG | 18:01:16 316 [Router] Received a ACK for 0x2acf29d6, stopping retransmissions
DEBUG | 18:01:16 316 [Router] Delivering rx packet (id=0xb9d09657 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=2acf29d6 rxtime=1749146476 priority=120)
DEBUG | 18:01:16 316 [Router] Update DB node 0x67893018, rx_time=1749146476
DEBUG | 18:01:16 316 [Router] Forwarding to phone (id=0xb9d09657 fr=0x67893018 to=0x67893018, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=2acf29d6 rxtime=1749146476 priority=120)
DEBUG | 18:01:16 316 [Router] Module 'routing' considered
INFO  | 18:01:16 316 Tell client we have new packets 14
```

fifieldt, 2025-07-28:
To fix this, we'd need to look up

hasOpenEditTransaction

from AdminModule.cpp in the webserver loop

or have something in AdminModule notify something in WebServer or WifiApClient ... 

Alternately, we could perhaps check the wifi status with something like:

WiFi.status() == WL_CONNECTED;

or maybe try to use

conn.wifi.status.is_connected

fifieldt, 2025-07-28:
@philon- , do you think this will work: https://github.com/meshtastic/firmware/pull/7484 ... do you have the capability to test?

closed 2025-07-30 by thebentern