meshtastic/firmware #6990: [Feature Request]: Wifi Bluetooth Switch

roboraptor, 2025-06-08:
### Platform

Cross-Platform

### Description

Hi, I came across the problem of partialy travelling nodes.
When I connect it to wifi, there is no easy way to connect it back to bluetooth when I take it out.
So, I came across a little solution. 
Would it be possible to implement GPIO function that would set module to WiFi/BT whel Low/Hi?

Xaositek, 2025-06-09:
@roboraptor I think this idea has merit, I put it in the backlog for the moment to revisit at a later time

Xaositek, 2025-06-24:
@roboraptor Reviewing this as a potential include in the Action menu designed in Meshtastic 2.7 - but it's taking a software approach to a hardware ask. Would screen-based solution be acceptable here, if not, please let me know.

roboraptor, 2025-06-24:
Hi, screen based could be sexy on devices which support it. But here I have Heltec V3 (and my friends has those Heltecs V3 too) where the switch would be super cool to have.

Xaositek, 2025-06-24:
Thanks @roboraptor I will investigate some software updates, but I updated the title to reflect the ask here.