meshtastic/firmware #7012: [Bug]: XIAO nRF52840 only probes for GPS at 9600 baud

Sandman224, 2025-06-10:
### Category

Serial

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.10.9ce4455 Alpha

### Description

I am trying to use an m10q GPS unit from newbeedrone with my Seeed Studio XIAO nRF52840 meshtastic kit. I have enabled the GPS and defined the RX and TX pins correctly as 7 and 6 respectively, and wired them correctly. I have tried changing the default baud rate on the GPS module using u-center 2 to 9600 but it seems that too much data is sent by this GPS unit to comfortably work at 9600 baud although it did accept the setting when I changed it. U center reports that the baud rate is too low (shocker) at 9600 which is why I would like to run the GPS at 115200 but the initialization sequence is not attempting to probe at that baud rate to detect the GPS.

Looking through the GPS.cpp file, and some other logs that people have posted, I see that the startup sequence is supposed to probe for GPS at multiple baud rates. However I am not seeing this behavior in the device logs as it only ever reports it is probing at 9600 and fails to detect the GPS. I wonder if this is due to a board configuration that locks the GPS baud rate to 9600?  There is a L76K module that is sold by Seeed Studio specifically for use with the Xiao, which runs at 9600 baud.

### Relevant log output

```Shell
//\ E S H T /\ S T / C

DEBUG | ??:??:?? 1 Filesystem files:
DEBUG | ??:??:?? 1  prefs (directory)
DEBUG | ??:??:?? 1    channels.proto (72 Bytes)
DEBUG | ??:??:?? 1    device.proto (127 Bytes)
DEBUG | ??:??:?? 1    module.proto (94 Bytes)
DEBUG | ??:??:?? 1    config.proto (208 Bytes)
DEBUG | ??:??:?? 1    nodes.proto (1369 Bytes)
DEBUG | ??:??:?? 1  adafruit (directory)
DEBUG | ??:??:?? 1  bond_prph (directory)
DEBUG | ??:??:?? 1  bond_cntr (directory)
DEBUG | ??:??:?? 1 Power::lipoInit lipo sensor is not ready yet
DEBUG | ??:??:?? 1 Use analog input 32 for battery level
INFO  | ??:??:?? 1 Scan for i2c devices
DEBUG | ??:??:?? 1 Scan for I2C devices on port 1
INFO  | ??:??:?? 1 No I2C devices found
DEBUG | ??:??:?? 1 acc_info = 0
INFO  | ??:??:?? 1 S:B:88,2.6.10.9ce4455
INFO  | ??:??:?? 1 Build timestamp: 1748871614
DEBUG | ??:??:?? 1 Reset reason: 0x0
DEBUG | ??:??:?? 1 Set random seed 945446475
INFO  | ??:??:?? 1 Init NodeDB
INFO  | ??:??:?? 1 Load /prefs/nodes.proto
INFO  | ??:??:?? 1 Loaded /prefs/nodes.proto successfully
INFO  | ??:??:?? 1 Loaded saved nodedatabase version 24, with nodes count: 17
INFO  | ??:??:?? 1 Load /prefs/device.proto
INFO  | ??:??:?? 1 Loaded /prefs/device.proto successfully
INFO  | ??:??:?? 1 Loaded saved devicestate version 24
INFO  | ??:??:?? 1 Load /prefs/config.proto
INFO  | ??:??:?? 1 Loaded /prefs/config.proto successfully
INFO  | ??:??:?? 1 Loaded saved config version 24
INFO  | ??:??:?? 1 Load /prefs/module.proto
INFO  | ??:??:?? 1 Loaded /prefs/module.proto successfully
INFO  | ??:??:?? 1 Loaded saved moduleConfig version 24
INFO  | ??:??:?? 1 Load /prefs/channels.proto
INFO  | ??:??:?? 1 Loaded /prefs/channels.proto successfully
INFO  | ??:??:?? 1 Loaded saved channelFile version 24
ERROR | ??:??:?? 1 Could not open / read /prefs/uiconfig.proto
DEBUG | ??:??:?? 1 cleanupMeshDB purged 4 entries
DEBUG | ??:??:?? 1 Use nodenum 0x3b402071
DEBUG | ??:??:?? 1 Expand short PSK #1
DEBUG | ??:??:?? 1 Expand short PSK #113
[... 127 lines cut ...]
DEBUG | ??:??:?? 13 [GPS] FromRadio=STATE_SEND_FILEMANIFEST
DEBUG | ??:??:?? 13 [GPS] File: device.proto (127) bytes
DEBUG | ??:??:?? 13 [GPS] FromRadio=STATE_SEND_FILEMANIFEST
DEBUG | ??:??:?? 13 [GPS] File: module.proto (94) bytes
DEBUG | ??:??:?? 13 [GPS] FromRadio=STATE_SEND_FILEMANIFEST
DEBUG | ??:??:?? 13 [GPS] File: config.proto (208) bytes
DEBUG | ??:??:?? 13 [GPS] Trying $PDTINFO (Unicore Family)...
DEBUG | ??:??:?? 13 [GPS] FromRadio=STATE_SEND_FILEMANIFEST
DEBUG | ??:??:?? 13 [GPS] File: nodes.proto (1369) bytes
DEBUG | ??:??:?? 13 [GPS] FromRadio=STATE_SEND_FILEMANIFEST
INFO  | ??:??:?? 13 [GPS] Config Send Complete
DEBUG | ??:??:?? 13 [GPS] FromRadio=STATE_SEND_PACKETS
INFO  | ??:??:?? 13 [GPS] toRadioWriteCb data 0x2001c732, len 41
DEBUG | ??:??:?? 13 [GPS] New ToRadio packet
DEBUG | ??:??:?? 13 [GPS] PACKET FROM PHONE (id=0x6d0f13e4 fr=0x00000000 to=0x3b                                                    402071, WantAck=1, HopLim=7 Ch=0x0 Portnum=6 priority=70)
DEBUG | ??:??:?? 13 [GPS] Enqueued local (id=0x6d0f13e4 fr=0x00000000 to=0x3b402                                                    071, WantAck=1, HopLim=7 Ch=0x0 Portnum=6 priority=70)
DEBUG | ??:??:?? 13 [GPS] FromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 13 [GPS] Trying $PCAS06,1*1A (ATGM33xx Family)...
DEBUG | ??:??:?? 14 [GPS] Trying $PAIR021*39 (Airoha Family)...
DEBUG | ??:??:?? 15 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 15 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 16 [GPS] Trying $PMTK605*31 (MTK Family)...
WARN  | ??:??:?? 17 [GPS] No GNSS Module (baudrate 9600)
WARN  | ??:??:?? 17 [GPS] Give up on GPS probe and set to 9600
```

Sandman224, 2025-06-10:
Changing those lines to the following and recompiling allowed the GPS to be picked up! 
#define PIN_SERIAL1_RX D6
#define PIN_SERIAL1_TX D7
Not exactly sure how contribution works for this repo but I can open a pull request if that is deemed appropriate.

fifieldt, 2025-06-10:
Confirming that #define GPS_BAUDRATE 9600 in the variant.h will cause this.

Since this module is not always sold with a GPS, this should be changed. Standby.

Sandman224, 2025-06-11:
Awesome thanks for updating that! I presume this will be included in the next alpha release?

closed 2025-06-13 by thebentern