meshtastic/firmware #7076: [Feature Request]: GPS enable/disable option

zmechu, 2025-06-19:
### Platform

ESP32

### Description

Hello!
Heltec V3 has ability to define with ability to enable/disable GPS module, switching on/off module supply, according to: https://meshtastic.org/docs/hardware/devices/heltec-automation/lora32/peripherals/

To enable/disable GPS module, user button should be pressed 3 times, ans - as I see- there is not other way.

Feature request: **to add possibility to add option to enable/disable GPS via mobile app**, sth like that:
[image]

It allows to remote and easy control over GPS module supply. Thank you in advance!

b8b8, 2025-06-19:
Did you check position settings?

closed 2025-06-19 by thebentern

zmechu, 2025-06-19:
> Did you check position settings?
> 
> This is already a feature for the iOS and Android app.

OK, I got it. It is quite similar - but enable/disable whole GPS functionality, which also needs device restart. It will be nice to add option just control GPIO (PIN_GPS_EN ) pin.

Regards!