meshtastic/firmware #7078: [Feature Request]: Setting to Adjust Bluetooth TX Power

asimons04, 2025-06-19:
### Platform

ESP32

### Description

Specifically requesting this for the ESP32-based boards, but would it be possible to add a setting to configure the Bluetooth TX power?

When I was working with my ESP32-based T-Watch, keeping Bluetooth enabled for notifications drained it's little 300 mAh battery in about an hour.  One thing that helped was setting the BT TX power to its lowest setting since it only needed to transmit from my wrist to my pocket.

Would that setting be possible to add for ESP32-based boards?  My Heltecs absolutely chew through battery when connected over Bluetooth.  I can only speak anecdotally, but there doesn't seem to be anything low power about BLE on an ESP32.

Joseph-DiGiovanni, 2025-07-05:
I understand this does not address your feature request, but unless you absolutely need an ESP32, you can use an NRF52 based radio which uses somewhere around 10x less power. 300mah would get you well over a day with a constant bluetooth connection.

Xaositek, 2025-07-05:
As a short-term answer to this, Meshtastic 2.7 can toggle Bluetooth on/off in the Home frame. This may improve battery overall more than just low TX power depending upon your needs

asimons04, 2025-07-05:
Good to know.  Sadly, not really helpful in my use-cases.  Would prefer the device to remain connected to my phone continuously.  I rarely even put the devices to deep sleep.