meshtastic/firmware #7080: [Feature Request]: Re-enable Detection module for T-1000E to detect button press on GPIO pin #6

yomanwill, 2025-06-19:
### Platform

NRF52

### Description

https://github.com/meshtastic/firmware/pull/6584 disabled many modules on T-1000E. Re-enable the detection module to allow for "button press" detection on GPIO pin number 6. This reddit poster claimed it was working on Firmware 2.5.20. https://www.reddit.com/r/meshtastic/comments/1f765ae/comment/mym56qf/?context=3

miketweaver, 2025-06-19:
I don't think you can enable this via CLI in any way.

The firmware is build by excluding this module:
https://github.com/meshtastic/firmware/blob/5e921453240b2d1be6a85c2dbaa35b5f1b1493c4/variants/tracker-t1000-e/platformio.ini#L9

There are a lot of places where the code for the module is excluded when that is set. Here's an example:
https://github.com/meshtastic/firmware/blob/5e921453240b2d1be6a85c2dbaa35b5f1b1493c4/src/modules/Modules.cpp#L24-L25

If you can show me otherwise, that would be awesome. But as it stands, #6584 broke this use case where we were using the GPIO from the built in button.

garthvh, 2025-06-20:
Building a custom firmware is the right way to re-enable this feature.

AndreasGaufer, 2025-07-30:
I use and recommend the exact method described in https://github.com/meshtastic/firmware/issues/7080#issuecomment-2991982990 and was also irritated by the module hide in android.

I don't understand how to do the suggested logo-long-press workaround. What logo should be pressed??